xsens/xsens_mti_ros_node #1: mtdevice.py not support the arguments set out in the documentation

savot, 2015-08-04:
I think there is something wrong with the file mtdevice.py. Giving a look at the source I can not find where the arguments -q, -t are managed, and in fact when I try to use them I get "option -q not recognized"

arunvydhya, 2015-08-05:
Yes thanks for reporting this. The documentation will be updated. Please use the argument '-f' to change the output data rate and '-m' for getting the sensor to the required sensor mode instead.

guneycan, 2015-09-09:
mtnode.py giving error (No module named custom_msgs.msg) although mtdevice.py working and streaming. Any recommendation?

arunvydhya, 2015-09-09:
@guneycan what kind of settings is the device set to? The custom_msgs is in the checked-in package. Did you run the catkin_make?

guneycan, 2015-09-09:
Yes I run the catkin_make and did 'source dev/ttyUSB0! and chmod 777 /dev/tyyUSB0.
Settings are default. I did not change anything. I need IMU+GPS+barometer.

arunvydhya, 2015-09-09:
@guneycan Thanks for the feedback. if you are interested in IMU+GPS+barometer please try the following:

rosrun xsens_driver mtdevice.py -f 400 -m 1
This would set the device to publish all data at 400Hz.

you can then access your data from, 
/mti/sensor/sample 
or from the following topics
/mti/sensor/imu
/mti/sensor/pressure
/mti/sensor/gnssPvt

you can use rostopic echo [msg]
to view data. Note that sample msg contains rate-integrating quantities whereas imu contains rate quantities. See MTi User Manual for more details.
Hope this helps.

closed 2015-09-09 by arunvydhya