xsens/xsens_mti_ros_node #2: Problem to visualize GPS data

daminienrico, 2015-10-10:
I'm working with a rover that has inside a MTi-G-700 GPS/INS and it's connected to a PC embedded with Linux. I have installed the drivers and I did what is written in the README.txt,the node runs correctly but it doesn't publish the GPS data (all the data are set to 0.0); **how I can configure the MTI???** I'm sure the MTI works because I'm using MT Manager for Linux (the software suite provided by Xsens) and I see the data and the satellites.

arunvydhya, 2015-10-14:
Configure the device to be in sensorMode 1 with a frequency say 100Hz (for publishing sensor data)
rosrun xsens_driver mtdevice.py -m 1 -f 100

Run the mtnode

You should be able to see the list of topics published, you are interested in /mti/sensor/gnssPvt

Hope this solves your issue.

closed 2015-10-14 by arunvydhya

daminienrico, 2015-10-15:
The problem was the version of the firmware: I used FirmwareUpdater in the MTi Software Suite.