xsens/xsens_mti_ros_node #4: Timeout and covariance questions and solutions

xsensapps, 2015-10-16:
User reported cloning the repo and running the node and the following commands with the IMU hardware (MTI-300) -
rosrun xsens_driver mtdevice.py -m 2
rosrun xsens_driver mtdevice.py -f 80
rosrun xsens_driver mtnode.py

User Question #1)  The node crashes after a few seconds with error "mtdef.MTException: MT error: timeout waiting for message".

Explanation/Solution: The timeout error usually happens in a slow system or when there are lots of parallel processes running. You might be able to play with the ‘additionalTimeOutOffset’ parameter in the mtdef.py script to avoid the timeout.

User Question #2) The covariances for all data (orientation, angular vel., linear acceleration) are published as 0, instead we would like to use actual values or at least a way to set them from a launch file using parameters.

Explanation/Solution: The covariance are not output by the filter and so it is set to zero. For values available, you may adapt the mtdevice.py script to publish values you want.

rs1990, 2016-02-23:
Hello ,

I have been getting the same issue as well with my MTi G700. Is there a proper fix to this ?

rs1990, 2016-02-24:
This is the output i keep getting : 

````
husky-a200@huskya200:~/Husky_indigo_ws$ rosrun xsens_driver mtdevice.py -m 3 -f 50
MT: Write message id 0x30 (GoToConfig) with 0 data bytes: []
MT error: timeout waiting for message.
husky-a200@huskya200:~/Husky_indigo_ws$ rosrun xsens_driver mtdevice.py -f 400 -m 1
````

arunvydhya, 2016-05-26:
@rs1990 I suggest playing around with the 'additionalTimeoutOffset' parameter as part of the mtdef.py. Try it after increasing it to 10ms or even more and let us know if the issue still persists.

closed 2016-05-31 by arunvydhya