xsens/xsens_mti_ros_node #8: Issue with MTi 1-series dev kit drivers after firmware update

gnarang1024, 2016-05-12:
Hello, I've been using a slightly modified version of these drivers with a [MTi 1-series Development Kit](https://www.xsens.com/download/pdf/documentation/mti-1/mti-1-series_dk_user_manual.pdf) (with the MTi-3-8A7G6 sensor) for some time without any problem. 

I recently updated the firmware from FW 1.0.6 version to the lates, FW 1.0.12 version and now, when I launch the xsens_driver.launch file I'm seeing the following error:

> 
>  SUMMARY
> 
> ========
> 
> PARAMETERS
>  * /rosdistro: indigo
>  * /rosversion: 1.11.19
>  * /xsens_driver/frame_id: imu
>  * /xsens_driver/frame_local: ENU
>  * /xsens_driver/frame_local_imu: ENU
> 
> NODES
>   /
>     xsens_driver (xsens_driver/mtnode_new.py)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> core service [/rosout] found
> process[xsens_driver-1]: started with pid [25162]
> [WARN] [WallTime: 1462986141.530980] Cannot find value for parameter: ~device, assigning default: auto
> [WARN] [WallTime: 1462986141.531970] Cannot find value for parameter: ~baudrate, assigning default: 0
> [INFO] [WallTime: 1462986141.621873] Detected MT device on port /dev/ttyUSB0 @ 115200 bps
> [INFO] [WallTime: 1462986141.622076] MT node interface: /dev/ttyUSB0 at 115200 bd.
> [INFO] [WallTime: 1462986141.663337] Found parameter: ~frame_id, value: imu
> Traceback (most recent call last):
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 495, in <module>
>     main()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 491, in main
>     driver.spin()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 111, in spin
>     self.spin_once()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 448, in spin_once
>     data = self.mt.read_measurement()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 390, in read_measurement
>     mid, data = self.read_msg()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 120, in read_msg
>     if ord(waitfor())<>0xFA:
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 118, in waitfor
>     raise MTException("timeout waiting for message.")
> mtdef.MTException: MT error: timeout waiting for message.

I'm running these drivers in ROS Indigo and everything was running perfectly with the previous firmware! It seems that after the firmware update the raw messages from the IMU is not being read by the ROS drivers(or maybe they are not in the correct format?). 

I also tried testing the IMU in the MT software suite V4.5.4 provided by xsens [MT_Software_Suite_Linux_4.5.4](https://www.xsens.com/mt-software-suite/) with the latest firmware(FW 1.0.12). Even though I was able to visualize the data from the IMU, I was getting the following out on my terminal window:

> 
>  ./mtmanager 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(351) setLogLevel Log level switched to [ERROR] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(365) setDebugLevel Debugger output log level switched to [FATAL] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(207) writeFileHeader Journaller logging to /home/gautam/.local/share/Xsens/mtmanager/xda.log for xda
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(208) writeFileHeader Current log level is [ERROR] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(212) writeFileHeader Journaller is not using threaded logging
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(365) setDebugLevel Debugger output log level switched to [FATAL] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(207) writeFileHeader Journaller logging to /home/gautam/.local/share/Xsens/mtmanager/mtmanager.log for MT Manager
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(208) writeFileHeader Current log level is [ERROR] 
> 1462986440.721 <740B4780> [WRITE] /home/builder/.pulse2-agent/data/agents/linux64-agent2/recipes/396918878/base/src/journaller.cpp(212) writeFileHeader Journaller is not using threaded logging
> 1462986441.328 <3EBE4700> [WRITE] createMasterDevice id: 03880223 firmware version: 1.0.12 build 106 rev 52752
>  

Please let me know if you have any suggestion for me. I'll be happy to provide you with any other information that might be helpful. Thanks in advance!

arunvydhya, 2016-05-13:
Thanks for your feedback! Can you let us know which raw IMU data (message ID's) have you enabled and what rates? 
We will be testing this issue and will get back to you within the next 2 weeks.

gnarang1024, 2016-05-18:
Hi @arunvydhya, I've tried setting different rates (1,5,10,20,40,50,80,100) while launching the node. Previously(with the old firmware), I was running the sensor at 100Hz and did not face any issues. 

For the raw IMU data, I've tried running it with different output settings but with luck. Currently, I'm running it on Normal Mode with no Preset selected. The selected timestamps are Packet counter and Sample Time Fine; Orientation(Quaternion); Inertial Data - Delta q, Delta v, Rate of Turn, Acceleration and the Magnetic field is also selected as an output. 

Also, I'm not sure about their exact MIDs but I've changed that so they should be they default ones.

gnarang1024, 2016-05-25:
# Update1 :
Upon Xsens support team's suggestion, I further update the firmware to 1.0.14 (from 1.0.12). But now I'm getting the following error while using the ROS drivers:

> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: indigo
>  * /rosversion: 1.11.19
>  * /xsens_driver/frame_id: imu
>  * /xsens_driver/frame_local: ENU
>  * /xsens_driver/frame_local_imu: ENU
> 
> NODES
>   /
>     xsens_driver (xsens_driver/mtnode_new.py)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> core service [/rosout] found
> process[xsens_driver-1]: started with pid [17511]
> [WARN] [WallTime: 1464190090.612756] Cannot find value for parameter: ~device, assigning default: auto
> [WARN] [WallTime: 1464190090.613608] Cannot find value for parameter: ~baudrate, assigning default: 0
> [ERROR] [WallTime: 1464190090.742375] Fatal: could not find proper MT device.
> [xsens_driver-1] process has finished cleanly
> log file: /home/gautam/.ros/log/42f8ede0-228c-11e6-8c25-507b9da34719/xsens_driver-1*.log
> all processes on machine have died, roslaunch will exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done
>  

The latest problem is that MTi device is not being detected by the ROS drivers after the firmware update. Also, Following is my output configuration settings:

[image]

[image]

# Update2 :

After setting the default port address and the baud rate to /dev/ttyUSB0 and 115200 in the node, I'm able to detect the sensor but I'm seeing the same error I was seeing with the previous firmware:

> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: indigo
>  * /rosversion: 1.11.19
>  * /xsens_driver/frame_id: imu
>  * /xsens_driver/frame_local: ENU
>  * /xsens_driver/frame_local_imu: ENU
> 
> NODES
>   /
>     xsens_driver (xsens_driver/mtnode_new.py)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> core service [/rosout] found
> process[xsens_driver-1]: started with pid [4751]
> [WARN] [WallTime: 1464192014.500343] Cannot find value for parameter: ~device, assigning default: /dev/ttyUSB0
> [WARN] [WallTime: 1464192014.501176] Cannot find value for parameter: ~baudrate, assigning default: 115200
> [INFO] [WallTime: 1464192014.501330] MT node interface: /dev/ttyUSB0 at 115200 bd.
> Traceback (most recent call last):
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 495, in <module>
>     main()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 490, in main
>     driver = XSensDriver()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py", line 56, in __init__
>     self.mt = mtdevice.MTDevice(device, baudrate)
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 30, in __init__
>     self.auto_config()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 377, in auto_config
>     self.GoToMeasurement()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 184, in GoToMeasurement
>     self.write_ack(MID.GoToMeasurement)
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 156, in write_ack
>     mid_ack, data_ack = self.read_msg()
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 120, in read_msg
>     if ord(waitfor())<>0xFA:
>   File "/home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtdevice.py", line 118, in waitfor
>     raise MTException("timeout waiting for message.")
> mtdef.MTException: MT error: timeout waiting for message.
> [xsens_driver-1] process has died [pid 4751, exit code 1, cmd /home/gautam/catkin_ws/src/ethzasl_xsens_driver/nodes/mtnode_new.py __name:=xsens_driver __log:=/home/gautam/.ros/log/aa00028c-228f-11e6-93a8-507b9da34719/xsens_driver-1.log].
> log file: /home/gautam/.ros/log/aa00028c-228f-11e6-93a8-507b9da34719/xsens_driver-1*.log
> all processes on machine have died, roslaunch will exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done
> 

Is there something that I'm missing here? Any suggestion would be helpful. Thanks!

hardikrshah007, 2016-05-25:
I am also having the exact same problem and I would really appreciate if I can do something and get it running. 

Thank You

hardikrshah007, 2016-05-25:
UPDATE : Below is the error that i am getting with the sensor running in Ubuntu 14.04 , I have faced no problems in running it on Windows 

hardy@whiteWalker:~$ rosrun xsens_driver mtnode.py _device:=dev/ttyUSB0 _baudrate:=115200
[INFO] [WallTime: 1464202348.318781] Found parameter: ~device, value: dev/ttyUSB0
[INFO] [WallTime: 1464202348.320396] Found parameter: ~baudrate, value: 115200
[INFO] [WallTime: 1464202348.320667] MT node interface: dev/ttyUSB0 at 115200 bd.
Traceback (most recent call last):
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtnode.py", line 373, in <module>
    main()
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtnode.py", line 368, in main
    driver = XSensDriver()
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtnode.py", line 61, in __init__
    self.mt = mtdevice.MTDevice(device, baudrate)
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtdevice.py", line 23, in __init__
    writeTimeout=timeout)
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 261, in __init__
    self.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 278, in open
    raise SerialException("could not open port %s: %s" % (self._port, msg))
serial.serialutil.SerialException: could not open port dev/ttyUSB0: [Errno 2] No such file or directory: 'dev/ttyUSB0'

hardikrshah007, 2016-05-25:
UPDATE 3 : I also got this error 

hardy@whiteWalker:~$ rosrun xsens_driver mtdevice.py -m 1 -f 20
Traceback (most recent call last):
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtdevice.py", line 992, in <module>
    main()
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtdevice.py", line 947, in main
    mt = MTDevice(device, baudrate)
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtdevice.py", line 29, in __init__
    self.auto_config()
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtdevice.py", line 503, in auto_config
    self.GoToConfig()
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtdevice.py", line 209, in GoToConfig
    self.write_ack(MID.GoToConfig)
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtdevice.py", line 182, in write_ack
    mid_ack, data_ack = self.read_msg()
  File "/home/hardy/catkin_ws/src/xsens/xsens_driver/src/mtdevice.py", line 165, in read_msg
    MID.ErrorCodes[data[0]]))
KeyError: 41

arunvydhya, 2016-05-26:
We are working on it and get back to you asap.

arunvydhya, 2016-05-26:
@hardikrshah007 , @gnarang1024 thanks for your feedback. I have been able to reproduce this on a slower PC. It could be very well possible that the timeout parameter is not enough. The default is now set to (1/samplerate + 6ms). You can probably increase it to 10ms in the mtdef source file -->'additionalTimeOutOffset'. Suggest also to test with a higher baudrate if possible. This issue is not expected to be a FW dependent issue but we are carrying out further checks here. Let us know if this helps.

arunvydhya, 2016-05-27:
@hardikrshah007 , @gnarang1024  if you have troubles getting the device recognized follow the tip below. In the upcoming commits I will add them to the documentation

- The Mti1 (Motion Tracker Development Board) is not recognized.

    Support for the Development Board is present in recent kernels. (Since June 12, 2015).
    If your kernel does not support the Board, you can add this manually

    $ sudo /sbin/modprobe ftdi_sio
    $ echo 2639 0300 | sudo tee /sys/bus/usb-serial/drivers/ftdi_sio/new_id

    - The device is recognized, but I cannot ever access the device -

    Make sure you are in the correct group (often dialout or uucp) in order to
    access the device. You can test this with

        $ ls -l /dev/ttyUSB0
        crw-rw---- 1 root dialout 188, 0 May  6 16:21 /dev/ttyUSB0
        $ groups
        dialout audio video usb users plugdev

    If you aren't in the correct group, you can fix this in two ways.

    1. Add yourself to the correct group
        You can add yourself to it by using your distributions user management
        tool, or call

            $ sudo usermod -G dialout -a $USER

        Be sure to replace dialout with the actual group name if it is
        different. After adding yourself to the group, either relogin to your
        user, or call

            $ newgrp dialout

        to add the current terminal session to the group.

    2. Use udev rules
        Alternatively, put the following rule into /etc/udev/rules.d/99-custom.rules

            SUBSYSTEM=="tty", ATTRS{idVendor}=="2639", ACTION=="add", GROUP="$GROUP", MODE="0660"

        Change $GROUP into your desired group (e.g. adm, plugdev, or usb).

    - The device is inaccessible for a while after plugging it in -

    When having problems with the device being busy the first 20 seconds after
    plugin, purge the modemmanager application.

gnarang1024, 2016-05-27:
Hi @arunvydhya , increasing the timeout parameter solved the issue for me. Thanks a lot for your help!

closed 2016-05-31 by arunvydhya

hardikrshah007, 2016-06-03:
Hi @arunvydhya , the Rosnode for the IMU is working fine right now . Thank you very much for your help . I was out of town so couldn't get to the issue in order to respond.