xsens/xsens_mti_ros_node #10: Reset Device for Calibration

hardikrshah007, 2016-06-03:
Hi Everyone 

I am using the IMU Mti-3 (development kit) version in ROS. I am using the filter VRU which provides me with unreferenced Yaw. I am using the IMU data in AMCL planner for mobile robots. 

At the beginning when the robot is turned on, I would like to do the following things

1. Reset the Gyroscope and Accelerometer to zero in order to calibrate as I know that my robot is stationary
2. Reset the Heading for the Yaw reading to zero ( i don't know what that value would be in Quaternion form)

arunvydhya, 2016-06-06:
@hardikrshah007 The reset command when sent while using in the 'vru_general' filter profile reinitializes the filter which will initialize the yaw reading to zero.

arunvydhya, 2016-06-06:
@hardikrshah007 you can already use the predefined 'reset' message by using the -r options that should do the trick. However this would however put the device to factory defaults. For in-run use, the node can be easily tweaked to send in a 'ResetOrientation' message with the code for heading reset. This can be done in-run (measurement mode). This should work for your application needs.

closed 2016-06-06 by arunvydhya

cherishyuan, 2024-10-22:
> Hi Everyone
>
> I am using the IMU Mti-3 (development kit) version in ROS. I am using the filter VRU which provides me with unreferenced Yaw. I am using the IMU data in AMCL planner for mobile robots. [...]

May I ask how you solved this problem?

Steven-GH, 2024-10-23:
> May I ask how you solved this problem?

Hi @cherishyuan,

This ROS node has been discontinued. I recommend using the following ROS1/ROS2 driver instead, which provides more configuration options as well:
https://github.com/jiminghe/Xsens_MTi_ROS_Driver_and_Ntrip_Client

For further support, it is best to contact Xsens at https://www.movella.com/support/request-support