xsens/xsens_mti_ros_node #11: timeout: catch exceptions

fherreroc, 2016-06-20:
I have been facing timeout exceptions which make the node crash. 
   `mtdef.MTException: MT error: timeout waiting for message.` 
   From function `read_msg` in `mtdevice.py`
Could exceptions be caught so the node can keep running? 
I'm using a Xsens MTi-300 IMU. Timeouts happen from time to time (don't know why yet).
Behavior can be tested by manually setting `autoconf=False` and a small `timeout` value.

arunvydhya, 2016-06-27:
@fherreroc timeout issue can be solved taking the measures suggested in this thread - https://github.com/xsens/xsens_mti_ros_node/issues/8
If you are still having issues try stripping off the node code to the bare minimum (to the data ID's that is required for your application).

fherreroc, 2016-07-05:
Well, yes, setting a high enough timeout value can avoid the problem, but, how high the value should be to ensure the node won't crash? My suggestion was leaving the timeout at a reasonable value, **catch** the timeout exceptions to ignore those particular iterations, and continue running the node.

I haven't stripped off the code yet, because I don't know the minimal data that we will need from the sensor.

arunvydhya, 2016-07-11:
In the systems I have tried it in, 10ms worked the best.

closed 2016-08-15 by arunvydhya