xsens/xsens_mti_ros_node #14: Mti-1 Development board

hardikrshah007, 2016-07-27:
Hi I am trying to run the Mti1 (Motion Tracker Development Board). I got the everything setup and I use the following commands
scrubba2@scrubba2-desktop:~$ **rosrun xsens_driver mtdevice.py -m 3 -f 10**  Timeout defaults to 0.110s based on output settings.
Device intiated at 10 Hz
Timeout changed to 0.110s based on current settings.
MTi-1/2/3 device detected
Enabled publishing all filter estimates
Device configured at 115200 bps
scrubba2@scrubba2-desktop:~$ **rosrun xsens_driver mtnode.py _device:=/dev/ttyUSB0 _baudrate:=115200**
Traceback (most recent call last):
  File "/home/scrubba2/catkin_ws/src/xsens_driver/src/mtnode.py", line 373, in <module>
    main()
  File "/home/scrubba2/catkin_ws/src/xsens_driver/src/mtnode.py", line 367, in main
    rospy.init_node('xsens_driver')
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    _init_node_params(argv, name)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 186, in _init_node_params
    set_param(rosgraph.names.PRIV_NAME + param_name, param_value)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/client.py", line 504, in set_param
    _param_server[param_name] = param_value #MasterProxy does all the magic for us
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rospy/msproxy.py", line 148, in __setitem__
    self.target.setParam(rospy.names.get_caller_id(), rospy.names.resolve_name(key), val)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1233, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1587, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1273, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1301, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1448, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 975, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 835, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 797, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 778, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 111] Connection refused

However , the above problem doesn't occur if I use 

scrubba2@scrubba2-desktop:~$ rosrun xsens_driver mtdevice.py -m 2 -f 10

and then run 
rosrun xsens_driver mtnode.py _device:=/dev/ttyUSB0 _baudrate:=115200

but I want the angular velocity and angular acceleration as well from the IMU and hence want to use mode 3 instead of 1 and 2. ( -m 2 -f 10)

arunvydhya, 2016-07-28:
@hardikrshah007 , mode 3 would get you the filter estimates of orientation. Mode 2 or 3 is what you need to use to access the calibrated inertial quantities.

hardikrshah007, 2016-07-28:
I do not get any angular velocity or linear acceleration data on the rostopic  /mti/sensor/imu. 

I tried changing the output setting on the sensor using the Windows application of xsens but still i do not get any angular velocity or angular acceleration data . So, if you could please help me enable the outputs

hardikrshah007, 2016-07-28:
I re-tried running the sensor and it started outputting data . However, there is a strange problem that i am facing which had led to reinstall the ros node . 

So, every few sequence, I get zero values on orientation data . 
Every few sequence , I get zero values on angular velocity data and linear acceleration. 

This is pretty bad for the robot as the zero orientation confuses the robot and it leads to errors in robot localization. So, if you have any comments on it .

hardikrshah007, 2016-07-28:
Below is the data output from the IMU. I am running at 10hz and in mode 2. 

header: 
  seq: 1
  stamp: 
    secs: 593
    nsecs: 403200000
  frame_id: /mti/data
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00230155908503
  y: 0.00749364122748
  z: 0.00258749164641
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.3487893641
  y: 0.173586905003
  z: 9.7748003006
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 2
  stamp: 
    secs: 593
    nsecs: 423200000
  frame_id: /mti/data
orientation: 
  x: 0.00755830621347
  y: -0.0175144579262
  z: -0.0938632190228
  w: 0.995402395725
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 3
  stamp: 
    secs: 593
    nsecs: 503200000
  frame_id: /mti/data
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00321738678031
  y: 0.00864424742758
  z: 0.000773883133661
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.340160280466
  y: 0.177225887775
  z: 9.79922676086
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 4
  stamp: 
    secs: 593
    nsecs: 523200000
  frame_id: /mti/data
orientation: 
  x: 0.00755358627066
  y: -0.0175263434649
  z: -0.0938834026456
  w: 0.995400369167
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 5
  stamp: 
    secs: 593
    nsecs: 603200000
  frame_id: /mti/data
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00386847066693
  y: 0.00798434112221
  z: 0.00248546712101
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.34631088376
  y: 0.202132061124
  z: 9.84716796875
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 6
  stamp: 
    secs: 593
    nsecs: 623200000
  frame_id: /mti/data
orientation: 
  x: 0.00754797272384
  y: -0.0175202284008
  z: -0.0938727036119
  w: 0.99540156126
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 7
  stamp: 
    secs: 593
    nsecs: 703200000
  frame_id: /mti/data
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00554508017376
  y: 0.00460299290717
  z: 0.00163549382705
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.331039607525
  y: 0.182855010033
  z: 9.81619930267
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 8
  stamp: 
    secs: 593
    nsecs: 723200000
  frame_id: /mti/data
orientation: 
  x: 0.00757667422295
  y: -0.0175600424409
  z: -0.0938975811005
  w: 0.995398342609
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 9
  stamp: 
    secs: 593
    nsecs: 803200000
  frame_id: /mti/data
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00410964572802
  y: 0.00917031615973
  z: 0.00151655101217
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.350101798773
  y: 0.184179127216
  z: 9.84749317169
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 10
  stamp: 
    secs: 593
    nsecs: 823200000
  frame_id: /mti/data
orientation: 
  x: 0.00754293380305
  y: -0.0176943168044
  z: -0.0939378365874
  w: 0.995392382145
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
header: 
  seq: 11
  stamp: 
    secs: 593
    nsecs: 903200000
  frame_id: /mti/data
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.00149373570457
  y: 0.00609705736861
  z: 0.00344177288935
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.343338549137
  y: 0.161625862122
  z: 9.82672691345
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---

arunvydhya, 2016-08-01:
@hardikrshah007 I am unable to reproduce this on my side. Are you using the latest 1-series FW v1.0.16?

arunvydhya, 2016-08-01:
@hardikrshah007 Also looking at the seq # and the time stamps (ie nsecs) it looks like you are dropping data packets (around 8 of them every now and then). Is your additionalTimeOutOffset set to 0.1s?

hardikrshah007, 2016-08-01:
Yes it is set to 0.1 ... I had reinstalled the driver from github and I can
amend if required ...

arunvydhya, 2016-08-02:
For your needs, setting the additionalTimeOutOffset to 10ms would suffice. There is something else going on as there are many dropped packets in the message stream that you have published, reasons could be e.g. slow system, network issues. Try stripping off the ros-node code to the message ID's you require and update to the latest FW.

closed 2016-08-24 by arunvydhya