xsens/xsens_mti_ros_node #21: mtnode.py throws a bunch of errors

spacemanspiff0211, 2017-06-30:
Hey,
I am trying to use this ROS node with my MTi-10, but the ROS node throws a bunch of errors.
the output of `rosrun xsens_driver mtnode.py _device:=/dev/xsens _baudrate:=115200` is 
```

[INFO] [1498865061.446497]: Found parameter: ~device, value: /dev/xsens
[INFO] [1498865061.447545]: Found parameter: ~baudrate, value: 115200
[INFO] [1498865061.447719]: MT node interface: /dev/xsens at 115200 bd.
Traceback (most recent call last):
  File "/home/spacemanspiff/catkin_ws/src/xsens_mti_ros_node/src/xsens_driver/src/mtnode.py", line 373, in <module>
    main()
  File "/home/spacemanspiff/catkin_ws/src/xsens_mti_ros_node/src/xsens_driver/src/mtnode.py", line 368, in main
    driver = XSensDriver()
  File "/home/spacemanspiff/catkin_ws/src/xsens_mti_ros_node/src/xsens_driver/src/mtnode.py", line 61, in __init__
    self.mt = mtdevice.MTDevice(device, baudrate)
  File "/home/spacemanspiff/catkin_ws/src/xsens_mti_ros_node/src/xsens_driver/src/mtdevice.py", line 23, in __init__
    writeTimeout=timeout)
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 180, in __init__
    self.open()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 311, in open
    self._update_dtr_state()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 605, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 22] Invalid argument

```
I did check the file permissions , everything seems to be alright. The output of `ls -l /dev/xsens`
is `lrwxrwxrwx 1 root root 7 Jun 30 15:57 /dev/xsens -> ttyUSB0`

Thanks

arunvydhya, 2017-07-03:
can you try running the xsens_driver without any arguments? Is the user part of the 'dialout' group? you can find the steps needed in the TroubleShooting section.

raymond30031, 2017-07-12:
Hello,

Same issue for me and I am using a MTi-G-710.
I have double checked the permission (chmod 666 /dev/ttyUSB0) and tried it without arguments and the issue still persists.

arunvydhya, 2017-07-17:
We have not been able to reproduce this on our side on an Indigo , which version of ROS are you using?

spacemanspiff0211, 2017-07-17:
Hey @arunvydhya 
I am using ROS Kinetic on Ubuntu 16.04.

raymond30031, 2017-07-18:
same for me

arunvydhya, 2017-07-18:
We are having some issues with the node in Kinetic. We are looking into this.

arunvydhya, 2017-07-18:
Can you replace line 22/23 in mtdevice.py with the following and try it? We will commit the changes asap.

self.device = serial.Serial(port, baudrate, timeout=timeout,
writeTimeout=timeout, rtscts=True, dsrdtr=True)

arunvydhya, 2017-07-19:
@spacemanspiff0211  and @raymond30031  did it work?

spacemanspiff0211, 2017-07-19:
@arunvydhya : Yep .. That fixes the issue for me..

closed 2017-07-20 by arunvydhya

sgarciav, 2017-08-03:
Hello,
Seems like this issue was closed but the fix was never committed. Took em a while to find this forum and solve the issue.

arunvydhya, 2017-08-09:
the commit has been made. Thanks!