xsens/xsens_mti_ros_node #25: set up exclusive access to serial port

ibtaylor, 2017-10-05:
If another process accidentally opens the serial port, bad things happen.

https://github.com/xsens/xsens_mti_ros_node/blob/0fa186ad14003b76e4e52860cc722e9f3bd24388/src/xsens_driver/src/mtdevice.py#L19-L40

Can we add something like [stackoverflow](https://stackoverflow.com/a/19823120) to set up exclusive access to the serial port?

arunvydhya, 2017-10-11:
Hi Ian,
We have looked at it. As you can see in the comments section - the fcntl solution protects only if the other processes use this as well. We are not in control of 3rd party applications and so adding it to our source code (as a generic solution) does not really help. This issue will be looked at and see if we can provide something else. Thanks for your feedback.

ibtaylor, 2017-10-12:
I guess I'm wondering if adding it would hurt anything. Currently, if a 3rd party application does not use a locking mechanism and attempts to communicate with the device while the xsens driver is running, it wrecks the xsens driver and interferes with the success of the 3rd party application.

I'm currently unaware of any cons of adding it to the xsens driver. It at least makes it **possible** for other processes to determine if it is safe to communicate with the device while the xsens driver code is running. 

If you know of some cons of adding it, please let me know for my own understanding.

arunvydhya, 2017-10-20:
There is no con on adding it to the code. We encourage you to send in a pull request which we can process. Thanks!

closed 2017-10-20 by arunvydhya