xsens/xsens_mti_ros_node #28: Timeouts and message fragmentation

collinsc, 2018-01-19:
Hello I have been working with this sensor for a few weeks, and I am running into some synchronization issues with the driver.

I have the device calibrated to: 
- 115200 baud
- output:
  - orientation: 50hz
  - angular velocity: 50hz
  - linear acceleration: 50hz

Examining the output of the device in mtmanager, the messages are small enough where they fit in a standard message. 

The issue I am having is that I need to run the driver at any less than 50hz or it fragments messages like the image below:
/home/nvidia/Pictures/Screenshot from 2018-01-18 22-00-25.png

I am in a pretty constrained system, and I don't want to run it that fast, but I do need smooth output from the driver to make our sensor fusion work.

I tried slowing down the publishing rate in the sensor firmware, but what I noticed is that it fragments mti messages into two messages at the device level, which causes the same issue of splitting the data at the driver level.

What I would like to do with the device is run it at ~15 Hz, and have the 3 sets of data output in one message. Is there something I am missing, or something I can modify in the driver/firmware to make this work?

arunvydhya, 2018-01-23:
@collinsc give us until the end of the week to get back to you on this. Thanks!

collinsc, 2018-01-23:
No problem! Thanks for the heads up.

arunvydhya, 2018-01-26:
We had a detailed look at this and we were not able to reproduce this on our side. We checked the driver logic for the outputs you have chosen and the driver should publish it in a single ROS message - so the issue is quite strange and probably system specific. 

We try to keep our driver updated and compatible to our FW changes and make sure that it works with new ROS releases. The primary focus of this driver is to kick-start ROS users in developing their application. In terms of code optimization or platform specific problems that users run into we advise them to look on appropriate ROS community forums for help. I am afraid we are unable to assist you further in this.

collinsc, 2018-01-27:
No problem thanks for looking in to it, if I track down what causes the crash, and it's something that can help others I'll make sure to post it as a response. Thank you.

arunvydhya, 2018-01-29:
thanks @collinsc

dan9thsense, 2018-02-27:
I am having the exact same problem.  It seems to be with the serial comm.
[image]
[image]

arunvydhya, 2018-03-07:
Hi @dan9thsense , we do not see this issue with our sensors. It could be a problem with the serial communication layer of the ROS node. To rule out any issues with the sensor, could you please try using the sensor with the Software Suite (MT manager) that is available here - https://www.xsens.com/mt-software-suite/ and see if you can reproduce it. Let us know. Thanks!

closed 2018-04-24 by arunvydhya