xsens/xsens_mti_ros_node #29: run error in ros

qiangshi1, 2018-03-07:
The problem is :
When I input $rosrun xsens_driver mtdevice.py -m 2 -f 100 # publish sensor data,
the terminal showed this :
MT error: timeout waiting for message.
or this :
MT error: Ack (0x31) expected, MID 0x36 received instead (after 100 tries).
In another terminal,I input $rosrun xsens_driver mtnode.py 
it showed this :
[INFO] [1520392053.315689]: Found parameter: ~device, value: /dev/ttyUSB0
[INFO] [1520392053.321804]: Found parameter: ~baudrate, value: 115200
[INFO] [1520392053.323191]: MT node interface: /dev/ttyUSB0 at 115200 bd.
Traceback (most recent call last):
  File "/home/qiangshi/catkin_ws4imu/src/xsens_driver/src/mtnode.py", line 373, in <module>
    main()
  File "/home/qiangshi/catkin_ws4imu/src/xsens_driver/src/mtnode.py", line 368, in main
    driver = XSensDriver()
  File "/home/qiangshi/catkin_ws4imu/src/xsens_driver/src/mtnode.py", line 61, in __init__
    self.mt = mtdevice.MTDevice(device, baudrate)
  File "/home/qiangshi/catkin_ws4imu/src/xsens_driver/src/mtdevice.py", line 29, in __init__
    self.auto_config()
  File "/home/qiangshi/catkin_ws4imu/src/xsens_driver/src/mtdevice.py", line 535, in auto_config
    self.GoToConfig()
  File "/home/qiangshi/catkin_ws4imu/src/xsens_driver/src/mtdevice.py", line 209, in GoToConfig
    self.write_ack(MID.GoToConfig)
  File "/home/qiangshi/catkin_ws4imu/src/xsens_driver/src/mtdevice.py", line 182, in write_ack
    mid_ack, data_ack = self.read_msg()
  File "/home/qiangshi/catkin_ws4imu/src/xsens_driver/src/mtdevice.py", line 140, in read_msg
    raise MTException("timeout waiting for message.")
mtdef.MTException: MT error: timeout waiting for message.
Besides,there is no topic published.

arunvydhya, 2018-03-07:
This is quite similar to the issue we have addressed before - please follow the steps suggested in #8

baronep, 2018-04-24:
@arunvydhya Regarding the timeouts, would it be possible to print a warning using ROS_WARN_THROTTLE instead of killing the node when this happens? It seems that this would make people aware of the issue, but the node would still remain functional.

arunvydhya, 2018-04-24:
@baronep Sure, we are on it. We will be pushing our commits tomorrow. Is it something you can test and let us know? Thanks!

baronep, 2018-04-24:
@arunvydhya I must actually correct myself on my previous comment. The ROS_WARN_THROTTLE has been available for a long time, but rospy.logwarn_throttle is only available in ROS kinetic or later.

arunvydhya, 2018-04-24:
@baronep yes that is the roscpp equivalent. It was a minor code change to get the rospy.logwarn_throttle in there. I will push it tomorrow.
Thanks for the note for users on the indigo - will add this to the changelog.

baronep, 2018-04-24:
@arunvydhya Sorry, my point was that if you could avoid using logwarn_throttle in favor of just logwarn that would allow indigo users (like myself) to continue to use the driver

arunvydhya, 2018-04-24:
thanks @baronep , will do. I will notify you once the code is pushed so that you can test and give me feedback.

arunvydhya, 2018-04-25:
@baronep can you have a look? I have pushed the commits.

baronep, 2018-04-25:
@arunvydhya I can confirm that this fix works as expected ...

**Test Case**
Configure device using mtexplorer to output data at 100Hz.
Set additionalTimeOutOffset = 0.001 in mydef.py
Launch node an observe occasional timout warn messages
Verified that IMU data is still being published to topic