xsens/xsens_mti_ros_node #30: Intermittent MT error on startup

baronep, 2018-04-23:
I am experiencing an intermittent issue with the ROS driver where I get the following error message when I try to run mtdevice.py.
```
#> ./mtdevice.py
MT error: Ack (0x31) expected, MID  0x36 received instead (after 100 tries)
```

The issue usually starts when I unplug the device while its running and will not recognize when I plug it back in. The problem usually goes away when I try a different USB port on my computer. To reproduce the issue I usually alternate between unplugging the device and running ./mtdevice.py

I am running this on the following system ...
```
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.4 LTS
Release:	16.04
Codename:	xenial
```

arunvydhya, 2018-04-24:
@baronep I have not been able to reproduce on my side so I am not sure how to help you on this. Looking at the MID it looks like the device is not getting into the 'Config' mode and is instead continuously spitting data in the 'Measurement' mode. Can you share the product code for the device you have?

baronep, 2018-04-24:
@arunvydhya I am testing on the following device ... 
```
Product ID: MTi-30-2AG8G4
Device ID: 0368201C
Firmware: 1.7.2.MTI5030.XS.28 (According to the calibration certificate)
Connection type: USB
```

I have been having quite a few issues that I believe to be related to the timeout on the device. I changed the additional timeout parameter to 20ms in mtdef.py and some other connection issues (same as those seen in #29) went away. Do you know why this timeout would be occurring?

arunvydhya, 2018-04-24:
@baronep I have seen this occur when someone chooses a low baudrate or run the node on slow systems or using the node in combination with other high-load demanding processes in parallel. You can try to get rid of this issue by selecting and publishing only messages that you need for your application. Yes, timeouts can lead to a situation that makes the device not enter the config mode.

closed 2018-07-11 by arunvydhya