xsens/xsens_mti_ros_node #35: MTi-G-710 output mode: would it be possible to get filter and sensor data together?

zongyaojin, 2018-07-31:
Hello,

I have the MTi-G-710 sensor, and running it with the ROS package. I notice in the `xsens.yaml`, it can either be configured to collect sensor data or filter estimates data with the following:

```
# Output data mode
# 1: sensor data
# 2. sensor data w/ rate quantities
# 3: filter estimates
output_mode: 3
```
I was wondering if it's possible to collect the sensor and filter estimates data together? If so, how would it do it? Thanks!

arunvydhya, 2018-08-01:
@zongyaojin 
Yes, you can easily modify the script to suit your application needs - if you modify the mtdevice.py with contents from line 418-516, shuffle and reorganize a bit you should be able to get both sensor and filter estimates.

zongyaojin, 2018-08-01:
Gotcha! Thanks a lot!

zongyaojin, 2018-08-01:
Hi @arunvydhya ,

Thank you for your reply and I've solved my problem! 

But there is another thing I wanted to ask. So, the MTi-G-710 is a really good sensor, but our lab needs some wireless sensors like this. I saw on the XSENS website, there is the MTw Awinda. But I didn't find a Github repo which sets up Awinda with ROS. 

So, I was wondering if your company offers a similar ROS package for MTw Awinda?

arunvydhya, 2018-08-02:
We do not have a ROS package for MTw Awinda but it has other features. Please look at our website and support forum for questions regarding MTw Awinda. thanks!

closed 2018-08-02 by arunvydhya