xsens/xsens_mti_ros_node #36: MTi-G-710 Launch Error :process has died

jessony, 2018-08-01:
I am using Mti-G-710 in ros-kinetic .And  I do the process as it says in https://github.com/xsens/xsens_mti_ros_node but when I do the step of "roslaunch xsens_driver xsens.launch",An error occured as following:

[xsens_mti-1] process has died [pid 6465, exit code 1, cmd /home/admin1234/catkin_ws3/src/xsens_driver/src/mtnode.py __name:=xsens_mti __log:=/home/admin1234/.ros/log/212d9eda-952c-11e8-a8d2-00216bec87bd/xsens_mti-1.log].
log file: /home/admin1234/.ros/log/212d9eda-952c-11e8-a8d2-00216bec87bd/xsens_mti-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I search the forum of base but haven't got  good solution.What should  I do to solve the problem?Thank you!
[image]

arunvydhya, 2018-08-01:
@jessony can you have a look at  this https://github.com/xsens/xsens_mti_ros_node/issues/34?
Also you can try updating the MTi-G-710's FW to the latest one.

zongyaojin, 2018-08-01:
Hi @jessony , I got your email. Below is what I did.

---

According to the [README.md](https://github.com/xsens/xsens_mti_ros_node/blob/master/README.md), these two lines solved my problem 

`$ sudo usermod -G dialout -a $USER
`
`$ newgrp dialout`

I just replaced `dialout` with my username `zongyao`, and you can try yours, hopefully that would work for you!

tiandaji, 2018-08-06:
I've encountered the same problem on G-710 firmware version 1.7.2 with @jessony  and @zongyaojin .
[image]
I tried the method zongyaojin proposed, but it did not work for me.
I've also tried to solve the group problem following another link:
http://www.360doc.com/content/14/0416/21/4683309_369568572.shtml​
which is "sudo usermod -aG dialout daji" and logout, then restart the session.
I'm sure that daji is already add in the group dialout, because when I type

sudo adduser daji dialout

The output is "The user `daji' is already a member of `dialout'."

However, I still get the same error.​

So I was wondering if the group problem is the cause of the ord error, 
and if it is just a matter of accessibility, why chmod can't do the job.
What's more, I can use MTManager to see the MT sensor data, which means
the PC already has the accessibility to the sensor.
From the source code, I think it is because an MTDevice class instance is not properly initialized,
thus no value for self.device.read().

Hope you @arunvydhya can give me some indications, and thanks very much in advance.

arunvydhya, 2018-08-06:
@tiandaji I can confirm that it is an issue with the FW v1.7.2 (with the 5th generation devices) --> a bug which is causing revert to fail. Can you update your device to the latest beta FW version 1.7.4 by following instructions on this page - https://base.xsens.com/hc/en-us/articles/208102825-Using-the-Firmware-Updater-with-a-beta-firmware

This should fix it, let us know if you still have issues!

tiandaji, 2018-08-07:
@arunvydhya I've updated the FW to v1.7.4, and the ROS node works well. However, the yaw's drift is very much, usually at a speed of 0.1 degree per second both indoor and outdoor without GNSS. For the version 1.7.2, this drift can be fixed using HighPerformance or GeneralMag filter, but for version 1.7.4, none of the five proposed filters work. Will you please give me some advice?

tiandaji, 2018-11-19:
@wujiang good for you, I managed to get MTi G710 work using GeneralMag filter after update of firmware and magnetic field calibration