xsens/xsens_mti_ros_node #38: MTi-G-710 filter estimates only getting orientation???

zongyaojin, 2018-08-02:
Hello,

I'm using MTi-G-710 with ROS. Two days ago it worked pretty well, but starting from yesterday, the filter estimates only getting orientation and missing velocity and position. I deleted the files on my laptop and downloaded again from the repo to make sure the code is right, but it's still having that problem.

Any suggestions about what causing the problem??? Below is my setup:

```
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /xsens_mti/baudrate: 115200
 * /xsens_mti/configure_on_startup: True
 * /xsens_mti/device: /dev/ttyUSB0
 * /xsens_mti/frame_id: xsens
 * /xsens_mti/odr: 1
 * /xsens_mti/output_mode: 3
 * /xsens_mti/use_rostime: True
 * /xsens_mti/xkf_scenario: 1

NODES
  /
    xsens_mti (xsens_driver/mtnode.py)

auto-starting new master
process[master]: started with pid [18472]
ROS_MASTER_URI=http://192.168.31.100:11311

setting /run_id to e13d7adc-9681-11e8-a682-34f39af35331
process[rosout-1]: started with pid [18485]
started core service [/rosout]
process[xsens_mti-2]: started with pid [18492]
[INFO] [1533234522.482688]: Found parameter: ~device, value: /dev/ttyUSB0
[INFO] [1533234522.483868]: Found parameter: ~baudrate, value: 115200
[INFO] [1533234522.484383]: MT node interface: /dev/ttyUSB0 at 115200 bd.
Timeout defaults to 1.010s based on output settings.
[INFO] [1533234522.870719]: Found parameter: ~configure_on_startup, value: True
[INFO] [1533234522.872696]: Found parameter: ~odr, value: 1
[INFO] [1533234522.877465]: Found parameter: ~output_mode, value: 3
[INFO] [1533234522.879254]: Found parameter: ~xkf_scenario, value: 1
[INFO] [1533234522.881285]: Found parameter: ~use_rostime, value: True
[INFO] [1533234522.881881]: Setting ODR (1) and output mode (3)
Timeout changed to 1.010s based on current settings.
MTi-G-700/710 (GNSS/INS) device detected.
Enabled publishing all filter estimates
Device configured at 115200 bps
Baudrate set to 115200 bps
Set to scenario: 1
[INFO] [1533234524.305939]: Found parameter: ~frame_id, value: xsens
[WARN] [1533234524.307501]: Cannot find value for parameter: ~frame_local, assigning default: ENU
[WARN] [1533234524.309245]: Cannot find value for parameter: ~frame_local_imu, assigning default: ENU
```

I also double checked the `mtdevice.py` and made sure that the `mtiMode' lines 448-461 are set up to get all the filter estimates data like below:

```
# Output configuration set based on the product ID and user specification	
		if (deviceIDProductMask[2] == XDIProductMask.MTi100Series) & (deviceTypeMask[2] == XDIProductMask.MTi700Device):
			print "MTi-G-700/710 (GNSS/INS) device detected."
			if mtiMode == 1:
				print "Enabled publishing all sensor data"
				data = mStf+mImuDq+mImuDv+mImuMag+mImuP+mGnssPvt+mGnssSat+mSw+mOrientationQuat
			elif mtiMode == 2:
				print "Enabled publishing all sensor data (rate quantities)"
				data = mStf+mImuGyr+mImuAcc+mImuMag+mImuP+mGnssPvt+mGnssSat+mSw+mOrientationQuat
			elif mtiMode == 3:
				print "Enabled publishing all filter estimates"
				data = mStf+mSw+mOrientation+mVelocity+mPosition+mHeight
			else:
				raise MTException("unknown mtiMode: (%d)."%	(mtiMode))
```

zongyaojin, 2018-08-02:
Could that be I'm in a GPS-denied area? Do the velocity estimates have anything to do with the GPS, or it's purely based on IMU's accelerometer?

arunvydhya, 2018-08-03:
@zongyaojin  velocity and position estimates are linked to GNSS updates. If you get into a GNSS denied environment after an initial GNSS fix then position and velocity estimates output will stop after 45-60s,  if the startup of the device is in a GNSS denied envrioment  you will no estimates of velocity and position will be available.

closed 2018-08-03 by arunvydhya