xsens/xsens_mti_ros_node #41: Xsens MTI-G-700-2A5G4 Drifting largely even when stationary

emil-jose, 2018-09-06:
Hi,
I am using Xsens MTI-G-700-2A5G4 to get imu data for a sensor fusion application and the imu measurements are observed to be drifting largely. I know the position estimation from IMU drift over time but the IMU data in my case makes me feel like the vehicle on which it is mounted is a high-speed train(I did the gravity compensation). I assumed Xsens MTI-G-700-2A5G4 is factory calibrated so this behavior is not expected I guess. Do I need to calibrate it? If yes could you point to some details on the same?  Also, can I get raw GPS measurements from the sensor itself using the driver? I haven't tried it outdoors.

StevenXsens, 2018-09-07:
Hello emil-jose,

I noticed that your questions are already being answered on [Xsens' BASE forum](https://base.xsens.com/hc/en-us/community/posts/360015912174-Xsens-MTI-G-700-2A5G4-highly-erroneous-data-from-IMU). If you have any questions specifically regarding Xsens' MTi ROS node, feel free to post them here. 

Raw GNSS Pvt messages are available via this driver as well. You can configure the MTi to output them by selecting output data mode 1 or 2 in the xsens.yaml configuration file. After running the node you can retrieve the data using:
`rostopic echo /mti/sensor/gnssPvt`

emil-jose, 2018-09-07:
I did set the mode to 1 and 2. It does not give me any GPS output. After trying it out, I wanted to switch to my previous ros driver (https://github.com/rpng/xsens_mti_ros_node), but it seems setting the mode did change something in the sensor itself. Now my previous driver does not seems to work properly.
Is there any way to reset it? and what could be the reason that 'rostopic echo' didn't show me any msg at all?

StevenXsens, 2018-09-10:
Hello emil-jose,

Are you getting any warnings or errors after using rostopic echo, or before that when launching the rosnode? Even without a GNSS fix, rostopic echo should return a list of 0-valued parameters. 

Upon startup the node takes the settings in the xsens.yaml file and uses them to configure the MTi. This means that its output configurations (output data mode/rate), filter profile and baudrate may change during the launch.

emil-jose, 2018-09-10:
The rostopic echo display nothing and launching the node . Even I tried outdoors and indoors. Its the same. I set the filter profile to 1 and changed the output data mode to 1 and 2 . Now I am getting this error

[WARN] [1536566817.813015]: timeout waiting for message
Traceback (most recent call last):
  File "/imu_ws/src/xsens_driver/src/mtnode.py", line 392, in <module>
    main()
  File "/imu_ws/src/xsens_driver/src/mtnode.py", line 388, in main
    driver.spin()
  File "/imu_ws/src/xsens_driver/src/mtnode.py", line 131, in spin
    self.spin_once()
  File "/imu_ws/src/xsens_driver/src/mtnode.py", line 148, in spin_once
    data = self.mt.read_measurement()
  File "imu_ws/src/xsens_driver/src/mtdevice.py", line 552, in read_measurement
    mid, data = self.read_msg()
  File "/imu_ws/src/xsens_driver/src/mtdevice.py", line 141, in read_msg
    if ord(c)<>0xFA:
TypeError: ord() expected a character, but string of length 0 found

StevenXsens, 2018-09-10:
Have you updated your MTi to run the lastest firmware version (v1.8.2) using our [Firmware Updater](https://www.xsens.com/mt-firmware/)? 

Also, are you using a direct USB connection or a serial to USB converter cable?

StevenXsens, 2018-09-10:
In that case I highly recommend first updating the firmware of your device to 1.8.2. If that doesn't solve the problem, you might want to check out the solution here: https://github.com/xsens/xsens_mti_ros_node/issues/36
(note that you are using a 4th generation device, so the latest firmware version is 1.8.2, in contrast to the 5th generation MTi's which are currently at 1.7.4.)

emil-jose, 2018-09-10:
I tried updating the firmware. It updated with message "unknown error while updating firmware". But anyway, now it seems to output the raw Raw GNSS Pvt messages and is working in my laptop. The current version it is showing it 1.8.2. When I plugged into the mobile platform, I got an error as in #36. I tried the solutions of adding myself to the group bu doesnt make any difference. I also tried "sudo chmod 777 /dev/ttyUSB0". The last suggestion in #36 is to update the firmware again to beta version.

StevenXsens, 2018-09-10:
Are you still seeing "TypeError: ord() expected a character, but string of length 0 found" upon launching the node? 

Your firmware version (1.8.2) is the latest available firmware version. There are no further beta firmwares available for your device.

emil-jose, 2018-09-10:
Yes, But only on my mobile platform(Segway). It works fine in my laptop.

emil-jose, 2018-09-10:
I was looking into the code for better understanding. I see a free acceleration output. Is free acceleration equivalent to gravity compensated? Is it possible to output both using the ros node.

wujiang, 2018-11-18:
You might have to do a magnetic field recalibration.