xsens/xsens_mti_ros_node #42: Usage of Filtered Orientation and Original Acceleration

tiandaji, 2018-10-29:
Hello @arunvydhya 
I am using xsens G710-2A8G4, and I would like to know if it is possible to get both filtered orientation and original acceleration using your python node.
I was able to get rostopic echo /mti/sensor/imu with the output like this:
header: 
  seq: 3010
  stamp: 
    secs: 1540801612
    nsecs: 818232059
  frame_id: "xsens"
orientation: 
  x: 0.0150714591146
  y: -0.00283857248724
  z: 0.968756854534
  w: 0.247538030148
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
However, whatever parameter I chose (rosed xsens_driver xsens.yaml, to change the output mode among 1, 2, 3) in the xsens.yaml, the accelerations remained zeros.
Will you please give me some indications?

tiandaji, 2018-10-30:
Furthermore, there are no output in output mode 3 for topic /mti/sensor/imu, which means even if the topic has output, the result will not be processed by the filter and remain original, which is not what I want. Thanks very much for all kinds of help @StevenXsens @arunvydhya

StevenXsens, 2018-10-31:
Hello Tiandaji, 

The ROS driver is merely a starting point for your application. It currently has three output data modes, available through the yaml configuration file:
1. (sensor data): this configures your MTi to output e.g. dq, dv and magnetic field data. 
2. (sensor data w/ rate quantities): this configures your MTi to output e.g. acceleration and rate of turn data. 
3. (filter estimates): this configures your MTi to output e.g. Euler angles, position and velocity. 

Next, there are several rostopic echo combinations available to visualize the data outputs. For example:
- /mti/sensor/imu will print quaternions, angular velocity and linear acceleration (if available)
- /mti/filter/orientation will print euler angles (if available)

In order to change the behaviour or add output modes, you can modify mtnode.py and mtdevice.py in the src folder to your needs.

hittingray, 2018-11-29:
Hello @tiandaji 

I know you asked a month ago, but I performed the modifications that Steven described above today. If you have not been able to get it working, here is what you need:

The line numbers below assume you have **unedited** files thus far:

In mtnode.py, on line 73:

Change `if output_mode not in [1, 2, 3]:` to `if output_mode not in [1, 2, 3]:`

In mtdevice.py, in the area of line 447, add the following lines under the section for your device:

```
elif mtiMode == 4:
    print "Custom"
    data = mStf+mImuGyr+mImuAcc+mImuMag+mSw+mOrientationQuat+mStf+mSw+mOrientation
```
If you are using a G710, this should be the first block.