xsens/xsens_mti_ros_node #45: Mti modes

HaroldMurcia, 2018-12-12:
Hi,
I'm using your node to configure a Mti-30 with ROS Kinetic. I'm interested in a implementation of a pose-estimation system based on the IMU.  I understand Mti-30 has a kalman filter for pitch/roll orientation. However if I choose Mti-Mode = 3 I have not information about accelerations, and when I chose Mti-Mode=[1,2] I have not information about estimated pitch/roll angles. ¿ How can I configure to obtain all the information even the free_accelerations ?

------------------------------ Line 476 of mtdevide.py
print "MTi-10/20/30 device detected"
                        if mtiMode == 1:
                                print "Enabled publishing all sensor data"
                                data = mStf+mImuDq+mImuDv+mImuMag+mSw+mOrientationQuat
                        elif mtiMode == 2:
                                print "Enabled publishing all sensor data (rate quantities)"
                                data = mStf+mImuGyr+mImuAcc+mImuMag+mSw+mOrientationQuat
                        elif mtiMode == 3:
                                print "Enabled publishing all filter estimates"
                                data = mStf+mSw+mOrientation
------------------------------

StevenXsens, 2018-12-13:
Hello Harold,

Yes, you can define a new mode, but you can also modify existing modes. For example, if you modify mtiMode == 3 to enable:
`data = mStf+mSw+mOrientation+mImuAcc+mImuGyr`

...then the MTi will also be configured to output Acceleration and Rate of Turn. You can receive the data using:
```
rostopic echo /mti/sensor/imu
rostopic echo /mti/filter/orientation
```

Another solution is to configure the MTi once using MT Manager before using the ROS node. Then, simply disable the configure_on_startup parameter in xsens.yaml. 

The ROS node is merely a starting point, so not all outputs have been defined. In order to configure and receive Free Acceleration using the ROS node, you will first need to define this output in the source code. A start would be: 
In mtdef.py, add the identifier to the XDIMessage class:
`FreeAcc = 0x00004030`
In mtdevice.py, add the Free Acceleration output to mode 3:
```
mFreeAcc = self.getMtiConfigBytes(XDIMessage.FreeAcc, rate_imu_period)
data = mStf+mSw+mOrientation+mFreeAcc
```
In mtnode.py, define a publisher and other related packet handling info. You can take the other output definitions as an example.

HaroldMurcia, 2018-12-14:
Thank you @StevenXsens ;
Now is working.

About Free accelerations: if the sensor is completely static, should accelerations be equal to zero?,  in my case they are not null: 
´´´
freeAccZ': -0.021139144897460938, 'freeAccY': -0.022071249783039093, 'freeAccX': -0.03868218883872032}. 
´´´
Do I need a calibration process?

StevenXsens, 2018-12-17:
Hello Harold,

The inertial sensor data (Rate of Turn, Acceleration) are never exactly 0 because of noise and [sensor bias](https://base.xsens.com/hc/en-us/articles/209611089-Understanding-Sensor-Bias-offset-). These inaccuracies will therefore also show up in the FreeAcceleration output. Although the MTi is constantly estimating its biases and aims to reduce their impact on the Free Acceleration output, some non-zero offset will always remain. What you can do is to apply some dynamics (i.e. rotations, accelerations) after you power up the device. This will allow the onboard filter of the MTi to better estimate the sensor biases.