xsens/xsens_mti_ros_node #47: Can MTi-10 IMU retrieve the magnetic field data using rostopic, like /imu/magnetic?

NTUwilliam, 2019-01-09:
Hi all

I'm wondering can the **MTi-10 IMU** output the magnetic field data as a **rostopic** like, for example: `/imu/magnetic` ?

If can, then will it output the **Magnitude** of the magnetic data (square root of (X*X+Y*Y+Z*Z)) or output the magnetic data as a **Vector** [X,Y,Z] ?

Thanks in advance!
@arunvydhya @fransklaver @savot

StevenXsens, 2019-01-09:
Hello @NTUwilliam,
Yes, the MTi-10 IMU can output magnetic field data. The data is presented as a vector (Mx, My, Mz). During factory calibration this vector is normalized to have a magnitude of 1. For more information, please refer to chapter 5.5.2 of the MTi User Manual: https://xsens.com/download/usermanual/ISM/MTi_User_Manual.pdf

If configured as an output (using the xsens.yaml configuration file or using MT Manager), the magnetic field data can be retrieved using:
`rostopic echo /mti/sensor/magnetic`

NTUwilliam, 2019-01-09:
@StevenXsens Noted with thanks!

I'll try it after I receive the MTi-10.

closed 2019-03-03 by NTUwilliam