xsens/xsens_mti_ros_node #52: MTi-10 rostopic output and magfieldmapper_gui

NTUwilliam, 2019-03-03:
Hi!

I just started to use MTi-10 IMU. And I'm wondering is it only able to output **4 kinds of data** ? Which are: 
1. /imu/mag 
2. /imu/dq 
3. /imu/dv 
4. /imu/time_ref

According to the [mti_driver ros document page](http://wiki.ros.org/xsens_mti_driver), I tried to echo some other topics like `rostopic echo /imu/data` or `/imu/acceleration`, but got nothing. 

Besides, I failed to start the `magfieldmapper_gui` executable to start **Magfield Mapper** in the magfieldmapper_linux-x64_version.tar.gz :(  
Is it something similar with the Magnetic Field mapping of the **inertial data mapping** in the **Xsens MT Manager** (as shown below)?

[image]

StevenXsens, 2019-03-04:
Hello William,
The MTi is capable of outputting more than the 4 types of data you have listed, but the MTi needs to be configured to do so. For this purpose, I recommend first using MT Manager (Device settings window, Output configuration tab). If desired, you can later use our programming examples to set desired output configurations through software commands. 

The Magnetic Field Mapper is a tool used to calibrate the magnetometers of your device for any static magnetic disturbances such as batteries or metal structures that are part of your setup. For more information, please see our [Magnetic Calibration Manual](https://xsens.com/download/usermanual/ISM/Magnetic_Field_Mapper_Documentation.pdf). 
If you are having troubles launching it, I recommend first reading the MFM.README document in the doc folder.

NTUwilliam, 2019-03-04:
@StevenXsens Thanks for replying!

I encountered with this while trying to do the calibration:
[image]

Does this mean that for MTi-10, the **calibration** can only be done by using the results of a previously recorded measurement (can not Use Motion Trackers ) ?

StevenXsens, 2019-03-04:
@NTUwilliam, my apologies for the confusion. I read past the fact that you are using an MTi-10. The MTi-10 is an IMU, which means that it is only capable of outputting calibrated and timestamped sensor data (i.e. acceleration, rate of turn, magnetic field). Sensor fusion outputs such as quaternion or free acceleration are only available with VRU, AHRS or GNSS/INS devices (i.e. MTi-20, MTi-30, MTi-G-710). 

The same holds for the magnetic field mapper: it can be used to recalibrate your magnetometers, but it requires onboard sensor fusion to do so. Therefore, the MFM is not available for your device.

NTUwilliam, 2019-03-05:
@StevenXsens Thanks.

> the MFM is not available for your device

By this, do you mean that the MTi-10 can not do the **recalibration** by any methods?

StevenXsens, 2019-03-05:
Yes, that is correct. In order to do a magnetometer calibration, the MTi has to measure the magnetic field for various orientations. As the MTi-10 cannot determine its own orientation, this is therefore not possible. The only way to recalibrate an MTi-10, is by returning it to Xsens for a recalibration. 
All sensor components (gyroscopes, accelerometers, magnetomers) are calibrated during factory testing and calibration. For the user there is currently no tool available that allows to recalibrate the gyroscopes and accelerometers.

NTUwilliam, 2019-03-07:
@StevenXsens 
Thanks!

By the way, the output of the magnetometer is in **arbitrary units** (`a.u.`), right? 
And can you confirm that for the magnetometer in MTi-10, one `a.u.` is approximately **40 μT** ?

StevenXsens, 2019-03-08:
@NTUwilliam The output of the magnetometers is indeed in arbitrary units. During in-house calibration (Enschede, the Netherlands), the magnetometers are calibrated such that the norm of the 3D magnetic field vector for any orientation equals 1. In fact, according to the current World Magnetic Model, this results in one a.u. being approximately equal to 49 uT.

NTUwilliam, 2019-03-08:
@StevenXsens 
Got it. Thanks a lot!

NTUwilliam, 2019-03-13:
@StevenXsens 
I got one more question. I browsed through the user manual of MTi-10 and I knew that it's unable to directly output the **`yaw`** information.
But I'm wondering is it possible to get the **`yaw`** information of the MTi-10 by some calculations of the available`magnetic field data` and `acceleration data`? 
I searched google and found some equations like:
1. `Heading (or yaw) =atan2( (-mag_y*cos(Roll) + mag_z*sin(Roll) ) , (mag_x*cos(Pitch) + mag_y*sin(Pitch)*sin(Roll)+ mag_z*sin(Pitch)*cos(Roll)) ) `
2. `yaw = 180 * atan (accelerationZ/sqrt(accelerationX*accelerationX + accelerationZ*accelerationZ))/pi`

Any hint or advice for this?
Thanks !

StevenXsens, 2019-03-14:
@NTUwilliam Yes, it is possible to determine the Yaw output based on the magnetic field itself. However, do note that the magnetic field may be influenced by hard and soft iron effects, such as magnets, batteries, metal structures, etc. This will disturb the Yaw you calculate. The MTi-30 for example can filter out these effects. 
Whether you can determine Yaw based on accelerometer data depends on your applications. For example, a drone can accelerate forward or sideways, while keeping the same Yaw. A car however, normally does not accelerate sideways. 

The MTi-10 does not have a world/earth-related coordinate frame, because it is not aware of its orientation and position on Earth. It only has sensor outputs based on its own XYZ sensor-fixed frame.

NTUwilliam, 2019-03-16:
@StevenXsens 
Noted with thanks!

NTUwilliam, 2019-03-18:
@StevenXsens 
Sorry for bothering you again. 
I've installed the MTi-10 on the robot in direction like this (**z-up**, **x-forward**,):
[image]

I found that the **vector** combination of the Magnetic **x and y direction** is roughly pointing to the North Geographical pole (South Magnetic pole). Is this correct / reasonable?

StevenXsens, 2019-03-18:
@NTUwilliam Yes, that is correct. You can simply use the X and Y components of the magnetic field for determining where North is. Do note however that the hard and soft iron effects of your robot will introduce a constant error in your magnetic North estimate, and any metal objects in the environment may introduce dynamic errors.

NTUwilliam, 2019-04-04:
@StevenXsens 
As I dig into the magnetic field theory recently, I got two more questions which I hope you or your colleagues could elaborate for me:
1. What the MTi-10 IMU measures is the **`Magnetic Flux Density`**(**B-field**), right? I suppose actually it is not only measuring the Magnetic field strength/intensity(**H-field**) but also measuring the Magnetization Field **M** since the `/imu/mag` output are in the units of **micro-Tesla(μT)**. 
2. Am I suppose to get the same value(apart from positive or negative) if I'm pointing the IMU at **exactly opposite direction**(180 degrees angle) at **exactly the same place** from the original one? Because it's just like turning the x/y axis to the opposite direction. But my test results failed to show this. 
For example, if I'm pointing the IMU at one direction I got the output value as **(-0.54, -0.46, 0.26)**, am I suppose to get **(0.54, 0.46, 0.26)** if I turned the IMU to the exactly opposite direction at exactly the same place? The actual value I got is **(0.67, 0.47, 0.29)** after turning 180 degrees. I did a couple of tests and found that after turning 180 degrees, the value in x/z directions would become bigger and the value in y direction would almost be the same(apart from positive or negative). 

Thanks a lot!!

StevenXsens, 2019-04-04:
@NTUwilliam 
1. This is correct.
2. Normally the X and Y components would indeed flip and the Z component should remain approximately the same. Note however that the magnetometer is not positioned exactly in the middle of the device; it is actually mounted near the label on the side of the MTi. Therefore, simply rotating the MTi on the spot will change the absolute position of the magnetometer.

NTUwilliam, 2019-04-07:
@StevenXsens 
Thanks a lot!

NTUwilliam, 2019-04-16:
@StevenXsens 
Sorry for bothering again.

With regards to the **resolution** of MTi-10, I'm thinking is there any possibility that the change in the magnetic field is _**smaller than**_ the resolution of the IMU so the IMU may give me almost the same output(can not distinguish these two different places)?

For example, I moved the IMU horizontally in the x-direction for 2cm, then the **/imu/mag output** are both (-0.33, 0.44, 0.32) before and after movement. The decimal from scale 3 to the end is floating(jumping) so I don't know how to judge whether the ambient magnetic field are the same before and after the 2cm movement.  

The resolution of MTi-10 is **0.25 mGauss(mG)**=**0.025 micro-Tesla(μT)**, right? If so, normally what's the **minimum distance** (only moving in one direction, keep the position in other two directions unchanged) MTi-10 need to distinguish two different places? 

Any hint or advice is appreciated.
Thanks!

StevenXsens, 2019-04-16:
@NTUwilliam 
I am not sure whether I understand your application correctly. The change in the measured magnetic field depends on the magnetic field you are in. Ideally, in a perfectly homogeneous Earth's magnetic field, you would not be able to distinguish between two places based on the magnetometers unless the MTi is rotated; the Earth's magnetic field will have the same direction and intensity. Only when the Earth's magnetic field is distorted locally, you will notice a difference in magnetic field between two locations. The intensity (and consequently the required resolution) however depends on the magnitude of the distortion and the distance between the MTi and the distortion. 

The resolution of the magnetometer components inside the MTi is indeed 0.25 μT.

NTUwilliam, 2019-04-17:
@StevenXsens 
Thanks for your clarification. 
If the resolution is indeed **0.25μT**, the **Magnetometer Specification information** on the user manual should be corrected(still shown as **0.25 mGauss**):
[image]

And:
> The only way to recalibrate an MTi-10, is by returning it to Xsens for a recalibration. All sensor components (gyroscopes, accelerometers, magnetomers) are calibrated during factory testing and [...]

By this, do you mean that the **magnetometer** inside MTi-10 can be re-calibrated by myself? If not, normally **after how much time** should the MTi-10 returned back to your company for a recalibration?

StevenXsens, 2019-04-17:
@NTUwilliam 
My apologies, I made an error there. I meant to say that the resolution of the magnetometer component is indeed 0.25 mGauss. The MTi User Manual is correct. 

The Magnetic Field Mapper is the tool to be used for recalibration of the magnetometers, especially when the MTi is integrated into an application. However, this requires the MTi to have an understanding of its orientation in space. The MTi-10 does not have this understanding, which means that it cannot be recalibrated with the Magnetic Field Mapper either. 

MTi's can be returned to Xsens for recalibration, but since the magnetometers are simply normalized to the local Earth's magnetic field and the noise levels and alignment are checked, there is no recommended recalibration period or metric for doing so. Usually MTi's are returned for recalibration because the gyroscopes or accelerometers are out of spec.

NTUwilliam, 2019-04-19:
@StevenXsens 
Understood. Sincerely thanks again!

NTUwilliam, 2019-05-07:
@StevenXsens 
Recently I'm considering buying MTi-20/30. But I have a few questions about the **yaw angle** which I hope you could clarify for me:

1. For the **MTi-30 AHRS**, the yaw output is referenced to True North. For the **MTi-20 VRU**, the yaw is relative yaw with the zero point being the yaw orientation when the filter was initialized (or when the last heading reset command was sent), right? 
    Does this mean that the yaw accuracy for MTi-30 is known but the yaw accuracy for MTi-20 is unknown? **Is the yaw accuracy the only difference between them**? 

2. What is the **exact form** will the **output of yaw angle of MTi-20/30** be? Is it like included in rostopic /imu/data ? Can you give me an example for this?

Thanks!

StevenXsens, 2019-05-07:
@NTUwilliam 

1. For the MTi-30 (AHRS), the Yaw is referenced to magnetic North based on the magnetometers, and aided by the gyroscopes. For the MTi-20 (VRU), the Yaw is unreferenced (starting at 0 degrees when powered up or after a heading reset). It is based on the gyroscopes and possibly aided by the magnetometers (AHS). 
Since the MTi-30 always is approximately correct, we can state a typical error of 1.0 deg for its Yaw output. The Yaw output of the MTi-20 will slowly drift away from the true value and therefore we cannot state a typical error for it. However, depending on the application, when [AHS ](https://base.xsens.com/hc/en-us/articles/211809465-Active-Heading-Stabilization-AHS-)is enabled, the drift over time can be as low as 3 deg/h. 
The Yaw output is the only difference between the two devices. Also, an MTi-30 can be configured to act as an MTi-20 using the so-called “vru_general” filter profile. 
2. The orientation output of the MTi-20/30 can be configured as:
a. Euler angles: Roll, Pitch and Yaw in degrees.
b. Quaternions
c. Rotation matrix
By default the orientation output is defined as the transformation between the sensor-fixed (XYZ) frame and the local earth-fixed (ENU) frame. For more info, refer to the [MTi User Manual](https://documentation.xsens.com/mti_user_manual). 
You can use for example rostopic echo /filter/quaternion or /imu/data to retrieve the orientation in ROS.

NTUwilliam, 2019-05-07:
@StevenXsens 
Noted with thanks!!

BTW, do you mean that the orientation output of the **MTi-10** is only available in the format of **Quaternions**, right? 
Because by `rostopic echo /imu/dq`, I can get **quaternion** data **x/y/z/w** from MTi-10.

StevenXsens, 2019-05-07:
@NTUwilliam
1. Correct. 
2. There is a difference between the quaternion output and the dq (delta quaternion) output. dq is simply the gyroscope data integrated over the last sample time. If you add up all these samples, you will obtain a dead-reckoned estimate of the orientation. It is available on all MTi models. 
The quaternion output is a sensor-fusion/filter based output that is only available on VRU, AHRS and INS devices. Thanks to the sensor fusion it is much more accurate than simply integrating gyroscope data.

NTUwilliam, 2019-05-08:
@StevenXsens 
Understood.
Thanks again for your patience and detailed answers!

NTUwilliam, 2019-05-31:
@StevenXsens 
Hi! I've received another MTi-10 and MTi-20. Now I have two MTi-10 and one MTi-20. 

So I'm wondering is it possible to launch **multiple xsens_mti_driver** in **one** laptop? Because I want to use these three sensors synchronously to collect data at the same time.

Thanks!

StevenXsens, 2019-06-13:
@NTUwilliam 
Sorry for the late reply - I was out of office the past two weeks. 

This is indeed possible by copying the existsing launch file and creating two additional launch files from it, i.e. "xsens_mti_node_2.launch" and "xsens_mti_node_3.launch". Open the new launch files and edit them to make sure that each node gets a different name. See also:http://answers.ros.org/question/151971/how-to-run-same-pakcage-twice-at-the-same-time/

NTUwilliam, 2019-06-17:
@StevenXsens 
Noted with thanks!!

WilliamWoo45, 2020-12-17:
@StevenXsens 
Hi! Sorry for bothering again.

Recently I plan to use the **Angular Velocity** data provided by the **MTi-10 IMU**. However I found that the `angular_velocity` contained in rostopic `/imu/data` is always zero (as shown below).
[image]

But in the **MT Manager User Interface**, there indeed is angular velocity data provided by the IMU (as shown below). 
[image]

So I'm wondering how could I obtain the angular velocity data by echo the rostopic?

Thanks!

Best Regards,
William

StevenXsens, 2020-12-17:
@WilliamWoo45 No problem!

It is important that the RateOfTurn output of the MTi is enabled. The angular velocity plot in MT Manager can also display dq or RateOfTurn**HR** data, after translating those data to deg/s. Could you check this on your side?

[image]

WilliamWoo45, 2021-01-26:
@StevenXsens

Thanks for your suggestion! It works.

Best Regards,
William

WilliamWoo45, 2021-02-17:
> Yes, that is correct. In order to do a magnetometer calibration, the MTi has to measure the magnetic field for various orientations. As the MTi-10 cannot determine its own orientation, this is [...]

@StevenXsens 
Hi!

Recently I found that my **MTi-10** needs to be re-calibrated because the measured magnitude (magnetic norm) is less than **0.6** in most cases.
I'm wondering can I re-calibrate the **MTi-10** by myself through the newest version of `Magfield Mapper`? I checked the [MT_Magnetic_Calibration_Manual](https://www.xsens.com/hubfs/Downloads/Manuals/MT_Magnetic_Calibration_Manual.pdf) and found at Section 4.2 (Off-line Magnetic Field Mapping) that maybe the **MTi-10** can be calibrated using a recorded `.mtb` file?

If **MTi-10** can not be re-calibrated by itself, may I use the `.mtb` file generated by another **MTi-20** to help with the **MTi-10** calibration process?

Thanks!

Best Regards,
William

StevenXsens, 2021-02-17:
@WilliamWoo45 I am sorry, but the Magnetic Field Mapper is simply not supported by the MTi-10 IMU (or vice versa). The Magnetic Field Mapper algorithm requires the magnetic field data as measured by the device itself during a calibration trial (hence why you cannot use another tracker), as well as the orientation estimates produced by the device for that same calibration trial (which are not produced by the MTi-10). Technically it could be possible if the Magnetic Field Mapper would produce orientation estimates for the recorded sensor data of an MTi-10, but this functionality is currently not implemented.