xsens/xsens_mti_ros_node #53: xsens_mti_ros not work on NVidia Jetson Nano

jianming1481, 2019-06-21:
Hello,

I have Xsens MTi-30 imu. I use it on Nvidia Jetson Nano with Ubuntu 18.04.02
Jetson Nano can recognize Mti-30 by typing lsusb

Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 008: ID 2639:0003 Xsens MTi-30 AHRS
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

However, i couldn't find it in /dev

So here are how i already tried:
1. build and install xsens_mt manually
2. download from dpkg xsens_mti_ros_node packge or build from source
3. sudo chmod 777 /dev/bus/usb/001/004
4. add the rule into /etc/udev/rules.d/

KERNEL=="ttyUSB*", ATTRS{idVendor}=="2639", ATTRS{idProduct}=="0003", MODE:="0777", SYMLINK+="xsense"

or

BUS=="usb", ATTRS{idVendor}=="2639", ATTRS{idProduct}=="0003", MODE="0666", SYMLINK+="xsens_mti"

All these method did not work!

So could you please help me with this problem?
Thank you!!!

StevenXsens, 2019-06-21:
Hello jianming,

For new designs we recommend using the ROS node that is included in the installation folder of the MT Software Suite, available for download at: https://www.xsens.com/mt-software-suite/
Use instructions and troubleshooting for this ROS node can be found at: http://wiki.ros.org/xsens_mti_driver

Could you let me know whether this solves your problem?

jianming1481, 2019-06-24:
no it doesn't work at all after run 
mtsdk_linux-64_2019.0.1.sh
and 
mfmsdk_linux-x64_2019.0.1.sh

BTW, xsens works on odroid

StevenXsens, 2019-06-24:
Hello jianming,

You should be able to execute the mtsdk_linux-64_2019.0.1.sh file using
`./mtsdk_linux-64_2019.0.1.sh`

After that, you will find a new folder xsens_ros_mti_driver.

jianming1481, 2019-06-24:
Hi Steven,

Thanks for your suggestions. 
However, I followed all the instructions from README.md
1. copy rospackage from mt sdk folder
2. build the libraries in xspublic
3. build the xsens_ros_mti_driver
4. add the rules to access xsens

when I launch 
roslaunch xsens_mti_driver xsens_mti_node.launch

show the error
No MTi device found

Note that I can see xsens by lsusb
but ls 
/dev|grep ttyUSB*
nothing show up

StevenXsens, 2019-06-25:
Hello jianming,

This is indeed quite strange. I have requested our software developers to have a look at this issue and hope to get back to you soon.

StevenXsens, 2019-06-25:
Hello jianming,

Could you check whether you have installed libusb-1.0?

jianming1481, 2019-06-25:
Hi Steven,

Really thanks for your reply.
I do install libusb-1.0-0-dev

jianming1481, 2019-06-25:
Sorry Steven,
I change the title. I can compile xsens_mti_ros on NVidia Jetson Nano.
Just couldn't find xsens in /dev. (I can see it in lsusb)

StevenXsens, 2019-07-03:
Hello jianming,

Our engineers have looked at the issue and they provided the following options:

1) You should make sure that the xsens_mt kernel driver is installed.
https://github.com/xsens/xsens_mt see here for the module and here for the installation steps:
```
$ git clone https://github.com/xsens/xsens_mt.git
$ cd ~/xsens_mt
$ make
$ sudo modprobe usbserial
$ sudo insmod ./xsens_mt.ko
```

After that, the driver should be loaded when connecting a device. It can be verified using the dmesg output which should print a line containing xsens_mt.ko

2) If you need to use libusb then you probably need to force compilation of the scanner with libusb.
For this you will need to recompile the libraries with the call 
`make HAVE_LIBUSB=1`

jianming1481, 2019-08-13:
Hey Steven,

Sorry for replying late,
I was busy for another project.
the solution you provide is work!
thanks a lot!!!

closed 2019-08-13 by jianming1481

mhaboali, 2020-01-24:
Hi @jianming1481 

I'm trying to install mti_ros_driver on Jetson TX2 board and I followed those steps you mentioned previously:

1. copy rospackage from mt sdk folder

> 2. build the libraries in xspublic

3. build the xsens_ros_mti_driver
4. add the rules to access xsens

But I got stuck at the second step, I didn't find the xspublic under lib folder. Have you encountered that scenario?

I hope you will help me with it, your help would be much appreciated!

StevenXsens, 2020-01-24:
Hello @mhaboali ,

Are you using the latest MT Software Suite v2019.**2.0** for Linux, available from our [website](https://www.xsens.com/software-downloads)?

mhaboali, 2020-01-27:
@StevenXsens 
Yes

StevenXsens, 2020-01-28:
Hello @mhaboali ,

The xspublic folder should be present at two locations:
- [mtsdk folder]/xsens_ros_mti_driver/lib
- [mtsdk folder]/public

Could you check this?

mhaboali, 2020-01-28:
Hello @StevenXsens 

I've re-downloaded the Linux suite again and this time I can see the xspublic folder as you described. 

Thanks so much for your patience!

jmgl0003, 2020-11-26:
Hello @StevenXsens

I am using a mti 680 g I have followed the steps in the thread but I cannot get it to associate the usb to ttyusb in the jetson nano. What I can do
Thanks so much

StevenXsens, 2020-11-27:
> 
> 
> Hello @StevenXsens
> 
> I am using a mti 680 g I have followed the steps in the thread but I cannot get it to associate the usb to ttyusb in the jetson nano. What I can do
> Thanks so much

Hello @jmgl0003 
We will continue troubleshooting on BASE: https://base.xsens.com/hc/en-us/community/posts/360010023540