xsens/xsens_mti_ros_node #54: ROS xsens regex error on Ubuntu 14

russelltankl, 2019-07-22:
I'm having an issue running the ros version for Xsens. I have the MTi-G-710 GNSS. I'm having a Ubuntu 14 machine, with gcc and g++ version 4.8. Building the application was not a problem. When I run roslaunch for the default launch file recommend in the readme, it throws an error:
"Terminate called after throwing an instance of 'std::regex_error' what(): regex_error"
 
I proceeded to upgrade my gcc and g++ version to version 6 and rebuild the Xsens ros package again. The same error still appears. 
 
The roslaunch works fine on my other machine with Ubuntu 16, although initially I had troubles with the package unable to detect my Xsens device. But I need the Xsens to work on the Ubuntu 14 machine.
 
I am using the latest xsens linux SDK and also recently updated the firmware.

Any help is really appreciated. Thank you!

StevenXsens, 2019-07-25:
Hello @russelltankl 
Sorry for the delay in getting back to you. 
Are you using the (deprecated) ROS node available on Github? Or are you using the newest version that is contained in the MT Software Suite (see http://wiki.ros.org/xsens_mti_driver)?

russelltankl, 2019-07-25:
Hi @StevenXsens 
Thanks for getting back to me! I used the latest version from the MT Software Suite.

StevenXsens, 2019-07-29:
Hello @russelltankl 

We need some more information regarding your setup and the error you are facing. Could you share the full error you receive, including the path where it occurs? 
Also, could you share more details regarding your setup (kernel, ROS version, etc..), as well as a confirmation that your current version (`gcc --version`) is the same as the one being used by ROS?