xsens/xsens_mti_ros_node #56: unable to get /filter/twist data

ajay1606, 2020-10-15:
Hello, I am working with Xsens MTi-670 model of IMU in Ubuntu 18.04 platform and trying to get /filter/twist data. There is no data in the topic name /filter/twist. Could anyone please suggest to me how to fix this issue? 

Regards,
AJay

StevenXsens, 2020-10-15:
Hello @ajay1606,

Please note that this ROS driver on Github is deprecated, it will no longer receive updates. The new and active ROS driver is part of the MT Software Suite for Linux, and its documentation can be found at http://wiki.ros.org/xsens_mti_driver. 

The /filter/twist data topic includes both Rate of Turn and Velocity data. In order to receive these data from your MTi you first need to configure it to output those, e.g. using MT Manager:
[image]

The Velocity data will not become available until you have achieved a GNSS fix.

ajay1606, 2020-10-15:
@StevenXsens Thanks for your quick response, Well currently am using xsens_mti_driver, downloaded and installed using mtmanager_linux-x64_2020.0. And you mentioned I have configured the device setting as shown in the image. But still twist data is not receiving in the topic /filter/twist.

1. device configuration
[image]

Result screenshot.
[image]

Looking forward to listening from you.

Thanks

StevenXsens, 2020-10-15:
Hello @ajay1606,

The Device Settings window shows that you have enabled a very large amount of data outputs, also at high rates (2000 Hz). Probably this is too much for your communication interface to handle, and as a consequence you may not be receiving all enabled data outputs. MT Manager will also display data overflow warnings with this configuration. 

I can recommend to disable AccelerationHR/RateOfTurnHR and to increase the baud rate of your MTi (e.g. to 921600 bps). Does that allow you to receive /filter/twist data?

ajay1606, 2020-10-16:
@StevenXsens , I have made changes to device configuration setting
 as suggested, but still /filter/twist is not able to receive.

1. Device setting
[image]

2. Sensor output configuration
[image]

Please could you confirm is there anything i am missing. 

Regards,
Ajay

StevenXsens, 2020-10-19:
Hello @ajay1606,

You have configured the RS232 baud rate at 921600 bps, but I assume that you are using the USB cable, is that correct? In that case you need to increase the UART baud rate.  

I also do not see the /filter/twist topic in your screenshots... It should look like this:
[image]

ajay1606, 2020-11-02:
@StevenXsens , Thanks it's working now. I have disabled unwanted data output. Now everything working fine.