xsens/xsens_mti_ros_node #58: Xsense Driver for ROS2

ROS-lover, 2020-11-30:
Hi,

I'm wondering if there are any plans to release ROS2 drivers as well.

Cheers,

StevenXsens, 2020-12-01:
Hello @ROS-lover,

We do indeed have plans to release a ROS2 driver, but development still needs to start for this. I will however register your request internally. Just to add to this, could you explain for what purpose/application you intend to use the ROS2 driver?

ROS-lover, 2021-01-21:
Hi there,

We are planning to use it for an indoor mobile robot that needs IMU for
localization.

Best,

ShepelIlya, 2021-01-27:
Hello, I'm intrested in ROS2 version too, any news?

StevenXsens, 2021-01-28:
We are currently waiting for a third party ROS 2 version to be published, and I will update you once that is available. 
Unfortunately I cannot yet provide details regarding an in-house developed ROS 2 driver.

olmerg, 2021-03-05:
Hello I am interested in ROS2 version, some news?

StevenXsens, 2021-03-08:
Hello @olmerg, unfortunately there are no updates yet that I can share.

StevenXsens, 2021-05-11:
Dear @olmerg, @ShepelIlya, @cassini-fly,

A third-party ROS2 driver has been published today, which is largely based on the official ROS1 driver by Xsens. You can find the ROS2 driver here: https://github.com/bluespace-ai/bluespace_ai_xsens_ros_mti_driver

vinci-xiao, 2022-06-16:
The third-party pkg(https://github.com/bluespace-ai/bluespace_ai_xsens_ros_mti_driver) seems to have some problems with recent modules. 
Will there be any official ros2 pkg released in the plan?

StevenXsens, 2022-06-17:
Hi @vinci-xiao,

Please see my response in the issue you linked to.

waltejon, 2022-10-31:
Dear all, I have an issue using the third-party ROS2 driver under Humble (see https://github.com/bluespace-ai/bluespace_ai_xsens_ros_mti_driver/issues/16). Does anyone have a tip how to solve this issue?

flabrosse, 2023-06-02:
Adding my voice.  We have to switch to ROS2 over the summer and looking at getting some MTi devices for our large all-terrain robots.  Not having a ROS2 driver would force us to look at other brands.

SilvioMueller, 2024-07-08:
Is there an update? Now that ROS Noetic EOL day is soon?