xsens/xsens_mti_ros_node #60: Testing on Ubuntu 14.04 with ROS Indigo - MT error: not an available XKF scenario

WilliamWoo45, 2021-01-20:
Dear @StevenXsens,

Hi!
I've successfully build the Xsens ros node on **Ubuntu 14.04** system with **ROS Indigo**.
However, when I try to run the node (`roslaunch xsens_driver xsens.launch`), there is an error saying that: "mtdef.MTException: MT error: not an available XKF scenario", as shown in the figure below.

[image]

Any solution for this?

Thanks and best regards,
William

WilliamWoo45, 2021-01-20:
Besides, I tried to download the software from your official website but also failed (as shown below, says not available). 

[image]

Is any of your older version software that can support the Ubuntu 14.04 system with ROS Indigo?

StevenXsens, 2021-01-20:
Hello @WilliamWoo45,

I believe that you are using an MTI-10 IMU, is that correct? If so, the MTi-10 IMU does not support any filter profiles (unlike the MTi-20 VRU and MTi-30 AHRS which have on-board sensor fusion algorithms) which explains the error you are getting. A quick fix should be to set _configure_on_startup_ to false in src/xsens_driver/config/xsens.yaml. 

Regarding the latest software download: Could you please check that page again? When I checked it just now, it said that the latest MT Software Suite is available for download.

WilliamWoo45, 2021-01-21:
Dear @StevenXsens,

Thanks for your prompt reply!

Actaully, I'm using `MTi-20 VRU`. However, when I set _configure_on_startup_ to false in src/xsens_driver/config/xsens.yaml, it has rostopic but there is no messages inside all the rostopic. It only works when I [set the _output_mode_ to 2](https://github.com/xsens/xsens_mti_ros_node/blob/16705507cfeaa227f763b6b689ce55ba78272fbe/src/xsens_driver/config/xsens.yaml#L31), and only the rostopic `/mti/sensor/imu`, `/mti/sensor/magnetic` have messages. The rostopic `/mti/filter/orientation`, `/mti/filter/position` and `/mti/filter/velocity` are empty. Any solution for this?

Besides, regarding your [previous answer](https://github.com/xsens/xsens_mti_ros_node/issues/52#issuecomment-490036170), I'm wondering why the MTi-20 is **NOT** able to output the referenced Yaw angle with respect to the magnetic North? Something as simple as the compass or the magnetometer inside the mobile phone can do that.

Further more, I downloaded the MTi software from your official website and unpacked it. I followed the installation instructions in the `MTSS.README` file, but neither did the "mtmanager" nor the "magfieldmapper_gui" is executable (as shown below). So how could I install and make them executable?
[image]

 

Thanks and Best Regards,
William

StevenXsens, 2021-01-21:
@WilliamWoo45 

If you are using the MTi-20, then it does indeed support a filter profile, but only profile number 43 (as the .yaml file also indicates). 

The MTi-20 does not compute estimates of position and velocity - only GNSS-enabled devices such as the MTi-G-710 can. Of course it is possible to use gyroscope and magnetometer data to estimate relative position and velocity, but this will suffer from significant drift over time due to integration errors. _/mti/filter/orientation_ should however be available, but only with output mode 3 (filter estimates). 

Of course the MTi-20 also features an internal magnetometer, from which you could directly calculate an estimate of the heading, but these heading measurements will not be reliable if magnetic distortions are present in the environment of the MTi. AHRS models such as the MTi-30 include algorithms that fuse the magnetometer data with the gyroscopes and accelerometers in order to produce a stable heading estimate, even in magnetically disturbed environments. The MTi-20 is a model that is designed for applications that do not require a (North-referenced) Yaw estimate. It is mostly used for indoor robotics, stabilization of platforms, marine applications, etc. 

Regarding the Linux programs; could you try to open a terminal and execute using _./mtmanager_? Note also that there are some dependencies that need to be installed, which are described in the doc folder.

danielschwartz3, 2021-03-27:
@StevenXsens I am unable to open mtmanager as well. When I try ./mtmanager in my linux terminal I get an error saying "-bash: ./mtmanager: cannot execute binary file: Exec format error"

StevenXsens, 2021-03-29:
@danielschwartz3 Sorry, there could be different things that can cause this error. Since this is not related to the ROS driver, could you contact Xsens Support via the following form, stating your MT Software Suite version, Linux OS version and what hardware you are using?
https://xsenstechnologies.force.com/knowledgebase/s/contactsupport

danielschwartz3, 2021-03-29:
@StevenXsens Thanks Steven, I'll check this out. I also have a question about the ROS driver.

I have my xsens MTi 670 connected by USB to a ROS robot. When I try to run the xsens_mti_node.launch file, I am getting the following output:
___________________________
... logging to /home/husarion/.ros/log/5edaf00a-90a8-11eb-b1b2-706655767e4d/roslaunch-rosbot-1140.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rosbot:46873/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /xsens_mti_node/pub_acceleration: True
 * /xsens_mti_node/pub_angular_velocity: True
 * /xsens_mti_node/pub_dq: True
 * /xsens_mti_node/pub_dv: True
 * /xsens_mti_node/pub_free_acceleration: True
 * /xsens_mti_node/pub_gnss: True
 * /xsens_mti_node/pub_imu: True
 * /xsens_mti_node/pub_mag: True
 * /xsens_mti_node/pub_positionLLA: True
 * /xsens_mti_node/pub_pressure: True
 * /xsens_mti_node/pub_quaternion: True
 * /xsens_mti_node/pub_sampletime: True
 * /xsens_mti_node/pub_temperature: True
 * /xsens_mti_node/pub_transform: True
 * /xsens_mti_node/pub_twist: True
 * /xsens_mti_node/pub_velocity: True
 * /xsens_mti_node/publisher_queue_size: 5

NODES
  /
    xsens_mti_node (xsens_mti_driver/xsens_mti_node)

ROS_MASTER_URI=http://master:11311

process[xsens_mti_node-1]: started with pid [1155]
[ INFO] [1617033929.028604465]: Creating XsControl object...
[ INFO] [1617033929.103092282]: Scanning for devices...
[ERROR] [1617033930.239625212]: No MTi device found.
[xsens_mti_node-1] process has died [pid 1155, exit code 255, cmd /home/husarion/ros_workspace/devel/lib/xsens_mti_driver/xsens_mti_node __name:=xsens_mti_node __log:=/home/husarion/.ros/log/5edaf00a-90a8-11eb-b1b2-706655767e4d/xsens_mti_node-1.log].
log file: /home/husarion/.ros/log/5edaf00a-90a8-11eb-b1b2-706655767e4d/xsens_mti_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
____________________________

I have spent a lot of time trying to figure out why the device is not being detected and I was wondering if you have any suggestions.

The device is connected, and when I run 'dmesg' I see that the device is detected:
__________
[    1.197742] usb 1-1: New USB device found, idVendor=05e3, idProduct=0610
[    1.197792] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    1.197833] usb 1-1: Product: USB2.0 Hub
[    1.198926] hub 1-1:1.0: USB hub found
[    1.199338] hub 1-1:1.0: 4 ports detected
[    1.306131] usb 3-1: new high-speed USB device number 2 using ehci-platform
[    1.476178] usb 1-1.1: new full-speed USB device number 3 using dwc2
[    1.586390] usb 1-1.1: New USB device found, idVendor=2639, idProduct=0300
[    1.586451] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.586508] usb 1-1.1: Product: Motion Tracker Dev. Board
[    1.586554] usb 1-1.1: Manufacturer: Xsens
[    1.586589] usb 1-1.1: SerialNumber: DB4IYUHS
[    1.592483] ftdi_sio 1-1.1:1.0: FTDI USB Serial Device converter detected
[    1.592744] usb 1-1.1: Detected FT-X
[    1.593559] usb 1-1.1: FTDI USB Serial Device converter now attached to ttyUSB0
_____________