xsens/xsens_mti_ros_node #61: Dev. Board(Mti-670): How to get acceleration and GPS data?

diennv, 2021-01-21:
Hi, i am using xda_cpp example from mtsdk with Mti-670 dev Kit. 
The below is my configuration.                
                configArray.push_back(XsOutputConfiguration(XDI_Acceleration, 100));
                configArray.push_back(XsOutputConfiguration(XDI_Quaternion, 100));
                configArray.push_back(XsOutputConfiguration(XDI_RawGyr, 100));
                configArray.push_back(XsOutputConfiguration(XDI_MagneticField, 100));
                configArray.push_back(XsOutputConfiguration(XDI_LatLon, 100));
                configArray.push_back(XsOutputConfiguration(XDI_AltitudeEllipsoid, 100));
                configArray.push_back(XsOutputConfiguration(XDI_VelocityXYZ, 100));

But i always only received the Orientation (XDI_Quaternion) data. How to configure to receive the others like: acceleration, gps ...?

Thanks,

StevenXsens, 2021-01-21:
Hello,

I recommend to enable XDI_RateOfTurn (calibrated gyroscope data) instead of XDI_RawGyr (raw gyroscope voltages). XDI_RawGyr cannot be combined with other outputs. 

Next, make sure that your output configuration is applied under the correct if statement:
[image]
It should be under ".isGnss()", because the MTi-670 is a GNSS/INS device.

Finally, the MTi will not start outputting position and velocity data until it has achieved a GNSS fix. Under a clear sky this may still take up to 30 seconds after powering up the MTi.

diennv, 2021-01-25:
Hi, Thank You for your kindly support.
I did the configure like this:

else if (device->deviceId().isGnss())
        {
                std::cout << "---------------------------Setting------------------------------" << std::endl;
                //configArray.push_back(XsOutputConfiguration(XDI_Quaternion, 100));
                //configArray.push_back(XsOutputConfiguration(XDI_Acceleration, 0));
                configArray.push_back(XsOutputConfiguration(XDI_RateOfTurn, 100));
                //configArray.push_back(XsOutputConfiguration(XDI_RawGyr, 100));
                //configArray.push_back(XsOutputConfiguration(XDI_MagneticField, 100));
                configArray.push_back(XsOutputConfiguration(XDI_LatLon, 100));
                //configArray.push_back(XsOutputConfiguration(XDI_AltitudeEllipsoid, 100));
                //configArray.push_back(XsOutputConfiguration(XDI_VelocityXYZ, 100));
        }
However, i didn't receive the acceleration yet. I also connected gps antenna to board and go out for test. 
I need the "angular velocity" and "acceleration" data for my works. Could you show me the configuration setting to achieve these data?
Thanks.

StevenXsens, 2021-01-25:
Hello,

I expect that it is because of this part of the example code:
[image]

packet.containsCalibratedData() will only return true if the packet contains Acceleration, RateOfTurn **and** MagneticField data. It should return true if you also enable XDI_MagneticField and XDI_Acceleration. Alternatively you could modify the code and use packet.containsCalibratedGyroscopeData() instead.

diennv, 2021-01-26:
Hi Steven,

Thank you for your helps. I did according to your instruction. 
I got acceleration data. How about "angular Velocity" data ?
I enable the configure "configArray.push_back(XsOutputConfiguration(XDI_VelocityXYZ, 100));".
But there are no data through this code:
 " if (packet.containsVelocity())
                        {
                                XsVector vel = packet.velocity(XDI_CoordSysEnu);
                                cout << " |E:" << vel[0]
                                        << ", N:" << vel[1]
                                        << ", U:" << vel[2];
                        }
"
Could you give me the hint?
Thanks.

StevenXsens, 2021-01-26:
Hello,

The MTi will not start outputting position and velocity data until it has achieved a GNSS fix. Under a clear sky this may still take up to 30 seconds after powering up the MTi. Have you connected the GNSS receiver and antenna to the MTi, and does it have a clear sky view? 

If you could share your current code then I can also have a look at it.

diennv, 2021-01-31:
Thank Steven,  
It may was affected by big building in my area. I used almost code from Xsens example code.
I will re-test when we go to clear area.
Thanks.

closed 2021-01-31 by diennv