xsens/xsens_mti_ros_node #62: Dev. Board(Mti-670): How to get linear_acceleration && angular_velocity in ROS driver

diennv, 2021-07-14:
Hi,
I am using the "xsens_mti_driver" with Mti-670 dev board in ROS.
I enabled the publish of acceleration and angular velocity information in config parameters file.
However i received only orientation data like the below.  How to change the config file to receive angular_velocity and acceleration ? 
---
header: 
  seq: 53265
  stamp: 
    secs: 1623904821
    nsecs:  31401642
  frame_id: "imu_link"
orientation: 
  x: -0.000409551139455
  y: -0.00898516736925
  z: -0.965376973152
  w: 0.260703444481
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.0
  y: 0.0
  z: 0.0
angular_velocity_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.0
  y: 0.0
  z: 0.0
linear_acceleration_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

-- -------------------------------The config file--------------------------------------

# Message publishers
pub_imu: true
pub_quaternion: true
pub_mag: false
pub_angular_velocity: true
pub_acceleration: true
pub_free_acceleration: false
pub_dq: false
pub_dv: false
pub_sampletime: false
pub_temperature: false
pub_pressure: false
pub_gnss: false
pub_twist: false
pub_transform: true
pub_positionLLA: false
pub_velocity: false

## Sensor standard deviation [x,y,z] (optional)
## This value is used to override the covariance matrix in sensor_msgs/Imu and
## sensor_msgs/MagneticField messages.
linear_acceleration_stddev: [0, 0, 0] # [m/s^2]
angular_velocity_stddev: [0, 0, 0] # [rad/s]
orientation_stddev: [0, 0, 0] # [rad]
magnetic_field_stddev: [0, 0, 0] # [Tesla]

StevenXsens, 2021-07-14:
Hello @diennv,

I believe that you are using the ROS driver that is contained in the MT Software Suite, is that correct? In that case, as also mentioned on http://wiki.ros.org/xsens_mti_driver, you will need to configure your MTi to output the necessary data messages. The easiest way to do this is by using the Device Settings window in MT Manager, see below screenshot. Alternatively you can use and modify the C++ example codes "example_mti_receive_data" in the MT SDK folder of your MT Software Suite. 

[image]

diennv, 2021-07-14:
Hi Steven,
Thank you for your quick reply. I will check and let you know the result.

Tks,

diennv, 2021-07-19:
Hi Steven,
I verified that your instruction was correct. Now, i can receive acceleration data & angular velocity.

Thank you so much,

closed 2021-07-19 by diennv