xsens/xsens_mti_ros_node #64: MTi-G-710-2A8G4 on ROS

ManChrys, 2021-08-05:
Hello guys,

i just bought MTi-G-710-2A8G4 and i want to use it on ROS environment.

I followed the directions from the [http://wiki.ros.org/xsens_mti_driver](url) to install the package.

It seems that is working fine but i would like to ask why angular_velocty and linear_acceleration are 0.0

[image]

ManChrys, 2021-08-05:
ok its fixed.

ManChrys, 2021-08-06:
ok i just fix it through MT Manager.

closed 2021-08-06 by ManChrys

flingok, 2023-09-06:
> ok i just fix it through MT Manager.

hi, I have the same issue, what is cause,could you please share it to me

ManChrys, 2023-09-06:
hello, did you try to find your module using MT Manager?