xsens/xsens_mti_ros_node #65: Interfacing MTi-G-710-2A8G4 with Livox mid 40

ManChrys, 2021-08-06:
Hello guys,

i would like to connect my MTi-G-710-2A8G4 with Livox mid 40.  Livox mid 40 interface is RS485.  According to manual if the number before A is 2 then the communication interface support RS232, USB, UART. But when i use MT Manager i saw that on log window

[image]

So i have to use a convert module TTL to RS485 or something else ?

StevenXsens, 2021-08-06:
Hello @ManChrys,

The MTi-G-710-2A8G4 is indeed a model that supports the RS232 and USB interface. The UART TTL interface is only accessible within the housing, which would be the "OEM" model (MTi-G-710-2A8G4-**O**). Products that support the RS485 interface have a "4" in front of the "A". That is why MT Manager also checks for devices connected over RS485. 

An external converter (UART TTL-to-RS485 or RS232-to-RS485) is indeed required in this case.

ManChrys, 2021-08-06:
@StevenXsens Ok i see!! Thank you for the fast response.

closed 2021-08-06 by ManChrys