xsens/xsens_mti_ros_node #67: Interface Diagram for Livox And Xsens Gnss/Ins

ManChrys, 2021-08-21:
Hello guys,

im trying to intergrate my MTi-G-710-2A8G4 with my Livox mid 40 laser scanner. On the diagram you can see my first try but it didnt work. The problem is that the xsens module doesnt appear on terminal when i tried `lsusb` but when i connect xsens directly to pc it worked normally.

[image]

If you can make a section on xsens site like these 

1 )https://base.xsens.com/knowledgebase/s/article/Interfacing-with-Ouster-Lidar-for-Xsens?language=en_US
2) https://base.xsens.com/knowledgebase/s/article/Interfacing-an-MTi-GNSS-INS-device-with-a-Velodyne-Lidar?language=en_US

about livox mid 40 it would be great. Although any help or tips or any diagram would be great.

Thanks in advance.

StevenXsens, 2021-08-23:
Hello @ManChrys,

This does not seem related to the ROS node... Could you contact Xsens Support for this? You can create a support case via [BASE](https://xsenstechnologies.force.com/knowledgebase/s/?language=en_US). Please then also provide more details on the multiplexer and the interface converters that are being used.

ManChrys, 2021-08-23:
Hello @StevenXsens 

I already an email with full details .

I have one more question . Using multiplexer converter when I run the `lsusb` command Mti device didn't get recogniZed. I think the reason behind this problem is the multiplexer .I don't know In my opinion the custom usb cable is not suitable for projects like mine because it's hard to sync gnss/ins with lidar and at the same time give data to pc.I think the one cable on xsens eshop with molex pins will be more useful and I think on this one all 9 pins are available something that doesn't happens on the custom usb cable has only 4 pins (VCC,GND,D+,D-) available .

ManChrys, 2021-08-23:
Hello @StevenXsens no I send it on info@xsens.com, but I'll send it on the email you said .

ManChrys, 2021-08-23:
@StevenXsens And last but not least. Is this cable [](https://shop.xsens.com/shop/mti-100-series/accessories/mti-100-series-accessories/ca-mp2-mti)

StevenXsens, 2021-08-24:
@ManChrys Yes, that would be the right cable to use.

ManChrys, 2021-08-24:
@StevenXsens When I'll try to connect the GNSS/INS using molex pins to pc through ttl to usb converter it will normally read it as xsens mti g710 or there is any chance to not detected as I said above when I used multiplexer ?

closed 2021-09-15 by ManChrys