xsens/xsens_mti_ros_node #68: Unable to get linear velocity and acceleration outputs on  MTi-30-2A8G4

kushal2507, 2021-11-25:
I am using an  MTi-30-2A8G4 on my mobile robot. When launching the node I can see the outputs in the /imu/data topic which is mostly the orienation but nothing on the /imu/dv or /imu/acceleration. 
Is this particular model not compatible to give linear velocity and acceleration? Or am I missing something in the param file?

All params have been set to true in the query.yaml file

Thank you!

StevenXsens, 2021-11-25:
Hello @kushal2507,

All MTi product support the dv and acceleration outputs. However you do need to configure your device to output these quantities, for instance using our MT Manager software. In MT Manager, open the Device Settings window, check the required outputs and click Apply:
[image]

kushal2507, 2021-11-26:
Thank you @StevenXsens this solves the issue!