xsens/xsens_mti_ros_node #70: Set covariance matrix for IMU Mti300 AHRS

Scarf195, 2021-12-21:
Hi,
I'm implementing an INS to obtain system state information using an Mti 300 device, integrating GNSS measurement when available using an Extended Kalman Filter.
I’m having trouble to set the covariance matrix for accelerometer and gyroscope as the sensor is mainly affected by bias and random noise available from datasheet.

Could you explain me how to set covariance matrix only for IMU measurements?

StevenXsens, 2022-01-03:
Hello @Scarf195,

Purely considering sensor noise levels, a 1 sigma standard deviation value for the acceleration and rate of turn outputs can be derived from the noise density values that are stated in the MTi 10/100-series user manual and the configured output data rate. Examples of that are shown here: https://xsenstechnologies.force.com/knowledgebase/s/article/RMS-noise-of-accelerometers-and-gyroscopes-1605869709686?language=en_US
The MTi 10/100-series user manual can be found at: https://mtidocs.xsens.com/mti-10-100-series-user-manual
You could use the squared values of the 1 sigma standard deviation to fill the diagonals of your covariance matrix.

In addition to noise levels, as you already suggested there are also other inaccuracies that are not easily captured using covariance matrices, such as bias instability over time and initial bias errors. I am afraid that I cannot support you in capturing those in your model. Usually these sensor biases are continuously estimated as part of the filter algorithms, as is also the case with the MTi-300.

narutojxl, 2022-01-07:
Hello @StevenXsens, 
I bought a MTi-G-710 and I don't find 1 sigma standard deviation param in [https://mtidocs.xsens.com/mti-10-100-series-user-manual](https://mtidocs.xsens.com/mti-10-100-series-user-manual), there are only noise density and in-run bias stability params about acc and gyro, except orientation.
[image]

[image]

[image]

In the aboving link, [https://xsenstechnologies.force.com/knowledgebase/s/article/RMS-noise-of-accelerometers-and-gyroscopes-1605869709686?language=en_US](https://xsenstechnologies.force.com/knowledgebase/s/article/RMS-noise-of-accelerometers-and-gyroscopes-1605869709686?language=en_US), it didn't tell how to calculate covariance from noise density and output data rate. Could you please explain how to do this, thanks your help and time very much!

StevenXsens, 2022-01-07:
Hello @narutojxl,

If you take the square of the standard deviation values, then you will end up with covariance values which you can use to fill the covariance matrix. 

The article does explain that the standard deviation values can be obtained from the noise density and the configured output rate. For instance, the accelerometer noise density is 60 µg/√Hz. At an output rate of 100 Hz, this means that the expected standard deviation of the acceleration data is 60*√100=600 µg = 0.006 m/s^2. 

This value however only incorporates sensor noise; it does not account for bias errors and bias instability. Those are usually estimated separately and therefore they are not part of the covariance matrix.

narutojxl, 2022-01-10:
Hello,  thanks @StevenXsens for your explanation :)

Scarf195, 2022-01-25:
Hi @StevenXsens, thank you very much for your reply! I've filled the matrix covariance as you suggest and now I'm developing an Error State Kalman Filter for bias estimation, improving my INS.
In this case, could I set my accelerometer bias error as a Gaussian, in particular N{0, σ} where σ is the in-run bias stability?
I hope I didn't make a mistake.

StevenXsens, 2022-01-26:
Hi @Scarf195, 
I am sorry, that is not something I can advise on. But it is important to understand that there are two parameters here:

- The initial bias error (0.05 m/s^2) represents the expected sensor bias that is present directly after power-up. Typically this value changes very slowly during the product's lifetime. A newly produced MTi will have a bias error that is very close to 0 because of the factory calibration. An MTi that is a couple years old can show a power-up sensor bias of +-0.05 m/s^2. 
- The in-run bias stability (15 ug) represents the stability of the sensor bias while the unit is powered up. Even if your filter were able to perfectly estimate the initial bias error at power-up, this instability would still cause measurement errors, resulting in drift in case  you are integrating gyroscope data.