xsens/xsens_mti_ros_node #73: xsens MTi-300 device not found

Amit10311, 2022-06-13:
Hello, 

I would like to mention that I can't find the sensor when using ros but when I am using `mtmanager` I found it. 

```
[ INFO] [1655136331.873229227]: Found device ID parameter: 037003F4.
[ INFO] [1655136331.873438885]: Found port name parameter: /dev/ttyUSB0
[ INFO] [1655136331.873450081]: Scanning port /dev/ttyUSB0 ...
```
My system is ubuntu 18.04 and ros Melodic. How can I solve this problem? Thank you!

StevenXsens, 2022-06-14:
Hi @Amit10311,

Are you using this (deprecated) ROS node on Github or the one that is included in the Xsens MT SDK?  
After plugging in the MTi, can you open a terminal and share the responses to the _dmesg_ and _lsusb_ commands?

Amit10311, 2022-06-14:
Hello,

Yes, This one helped it. It's working perfectly. 

 **"The device is recognized, but I cannot ever access the device." on the following page:
http://wiki.ros.org/xsens_mti_driver**

```
$ sudo usermod -G dialout -a $USER

$ newgrp dialout
```

Thank you for the quick response and solution.

closed 2022-06-14 by Amit10311