xsens/xsens_mti_ros_node #74: Xsens MTIG-710 roslaunch error... process dies

ahar, 2022-07-04:
Hi,

Please find the attached image. I followed the troubleshooting method explained in the readme file. But, the error sustains.

Your help is much appreciated.

regards,

[image]

StevenXsens, 2022-07-04:
Hello @ahar,

It looks like the "xkf_scenario" number you provided in the xsens.yaml configuration file is invalid for the MTi-G-710. Its value should be in the range of 1 to 5. 
Anyway, I highly recommend using the latest ROS (actively maintained) ROS driver which is described here: http://wiki.ros.org/xsens_mti_driver

ahar, 2022-07-07:
Hi @StevenXsens 
Thanks, that issue is resolved. With the latest software from [here](https://www.xsens.com/software-downloads) I am able to use .mtmanager to view the data generated by the sensor. But, I did not find any link for the ROS wrapper on the wiki page that you referred. Can you please suggest on this?

StevenXsens, 2022-07-07:
Hi @ahar,

Have you downloaded the MT Software Suite for MTi products (2021.x)? Then, extract the downloaded installation package, execute the mtsdk_linux-x##_####.#.sh and you will find the xsens_ros_mti_driver folder.

StevenXsens, 2022-07-12:
Hi,

I have the feeling that you are not using the latest xspublic folder contents because the latest one includes a file mti7_mti8device.cpp.o instead of mti7device.cpp.o. Could you try simply copying the complete folder _xsens_ros_mti_driver_ from your MT SDK (v2021.4) to the _src_ folder of your ROS workspace? 

If the problem still shows up, please copy the full terminal trace and send it to support@xsens.com together with your system info (ROS version, Ubuntu version, etc.) such that our support team can have a look at it.

ahar, 2022-07-12:
Very interestingly, I tried everything inside one of my docker containers. It worked. Also, I checked that `mti7_mti8device.cpp` existed inside the `xspublic` directory while I tried from within my ros workspace outside the container. It might have happened due to a permission issue, although I am not sure. Because I manually checked the `rw` permissions to each file.

Anyway, thank you for the support.

closed 2022-07-14 by ahar