xsens/xsens_mti_ros_node #75: Linking problem occurs when installing MT Software Suite

Taeyoung96, 2022-09-28:
Hi, @StevenXsens  
I want to use MTi-100 IMU on my desktop.  
My OS version is Ubuntu 18.04.  

According to [ROS documentation](http://wiki.ros.org/xsens_mti_driver), I download latest MT software suite.  
When I command `sudo ./mtsdk_linux-x64_2021.4.sh`, I got following errors.  
```
Installing 'Xsens MT SDK'
Enter the installation directory [/usr/local/xsens]
> 
#142~18.04.1-Ubuntu SMP Thu Sep 1 16:25:16 UTC 2022
Start using by adding -L/usr/local/xsens to your linker options
Or register systemwide using ldconfig /usr/local/xsens
```  

How could I fix this problem?  

I tried to add some path to ldconfig, but I don't know which absolute path to add.  
In my `/etc/ld.so.conf.d`, the following list exists.  
```
/usr/local/lib/x86_64-linux-gnu
/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu
/usr/local/lib
/usr/lib/x86_64-linux-gnu/libfakeroot
/usr/local/lib/i386-linux-gnu
/lib/i386-linux-gnu
/usr/lib/i386-linux-gnu
/usr/local/lib/i686-linux-gnu
/lib/i686-linux-gnu
/usr/lib/i686-linux-gnu
```

I'd appreciate it if you could give me an answer.  

Thanks,

StevenXsens, 2022-09-28:
Hi @Taeyoung96,

This message will not disappear, but it is basically a reminder telling you to do the following:

- Create a new file _xsens.conf_ in directory _/etc/ld.so.conf.d_   
- Add the following line to it: _/usr/local/xsens/lib_
- Rebuild the cache using _sudo ldconfig_
- Check whether it was successful with _ldconfig -p | grep xsens_. You should see something like this:
        libxstypes.so.2021 (libc6,x86-64) => /usr/local/xsens/lib/libxstypes.so.2021
        libxstypes.so (libc6,x86-64) => /usr/local/xsens/lib/libxstypes.so
        libxsensdeviceapi.so.2021 (libc6,x86-64) => /usr/local/xsens/lib/libxsensdeviceapi.so.2021
        libxsensdeviceapi.so (libc6,x86-64) => /usr/local/xsens/lib/libxsensdeviceapi.so
        libxkf700_filter.so.6 (libc6,x86-64) => /usr/local/xsens/lib/libxkf700_filter.so.6
        libxkf700_filter.so (libc6,x86-64) => /usr/local/xsens/lib/libxkf700_filter.so
        libxkf3i_filter.so.5 (libc6,x86-64) => /usr/local/xsens/lib/libxkf3i_filter.so.5
        libxkf3i_filter.so (libc6,x86-64) => /usr/local/xsens/lib/libxkf3i_filter.so
        libxkf3hm_filter.so.0 (libc6,x86-64) => /usr/local/xsens/lib/libxkf3hm_filter.so.0
        libxkf3hm_filter.so (libc6,x86-64) => /usr/local/xsens/lib/libxkf3hm_filter.so
        libxkfcore_filter.so.1 (libc6,x86-64) => /usr/local/xsens/lib/libxkfcore_filter.so.1
        libxkfcore_filter.so (libc6,x86-64) => /usr/local/xsens/lib/libxkfcore_filter.so

Taeyoung96, 2022-09-28:
@StevenXsens   
Thanks for your reply.  
With your help, I succeeded in downloading the MT software suite. :smile:   

However, I encounter another problem.  

The device seems to be recognized, but I don't think I can get the IMU data.  
When I execute `roslaunch xsens_mti_driver display.launch`,  
The results are shown below.  

```
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://taeyoung-cilab:42647/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /xsens_mti_node/pub_acceleration: True
 * /xsens_mti_node/pub_angular_velocity: True
 * /xsens_mti_node/pub_dq: True
 * /xsens_mti_node/pub_dv: True
 * /xsens_mti_node/pub_free_acceleration: True
 * /xsens_mti_node/pub_gnss: True
 * /xsens_mti_node/pub_imu: True
 * /xsens_mti_node/pub_mag: True
 * /xsens_mti_node/pub_positionLLA: True
 * /xsens_mti_node/pub_pressure: True
 * /xsens_mti_node/pub_quaternion: True
 * /xsens_mti_node/pub_sampletime: True
 * /xsens_mti_node/pub_temperature: True
 * /xsens_mti_node/pub_transform: True
 * /xsens_mti_node/pub_twist: True
 * /xsens_mti_node/pub_velocity: True
 * /xsens_mti_node/publisher_queue_size: 5

NODES
  /
    rviz (rviz/rviz)
    xsens_mti_node (xsens_mti_driver/xsens_mti_node)

ROS_MASTER_URI=http://localhost:11311

process[xsens_mti_node-1]: started with pid [14769]
process[rviz-2]: started with pid [14770]
[ INFO] [1664375521.214797799]: Creating XsControl object...
[ INFO] [1664375521.231425633]: Scanning for devices...
[ INFO] [1664375521.264140867]: Found a device with ID: 017828BD @ port: /dev/ttyUSB0, baudrate: 115200
[ INFO] [1664375521.264186657]: Opening port /dev/ttyUSB0 ...
[ INFO] [1664375521.319884915]: Device: MTi-100-2A8G4, with ID: 017828BD opened.
[ INFO] [1664375521.350122846]: Measuring ...
```

And the Rviz screen is like below,  
[image]

I want to use MT Manager, I change py path to `~/Downloads/MT_Software_Suite_linux-x64_2021.4/mtmanager/linux-x64/bin`  
I exectue `sudo ./mtmanger`.  

And the output is below.  

[image]

Nothing is being printed on the screen.  
Which part should I fix first?  

Thanks,

StevenXsens, 2022-09-29:
Hi @Taeyoung96,

This means that your MTi device is not configured to output the messages that are required by the ROS driver (or for displaying in MT Manager). You can enable the correct messages in the Device Settings menu of MT Manager: 
[image]

However, the above screenshot is for an MTi-**3**00 AHRS device which has onboard sensor fusion algorithms. The MTi-**1**00 which you are using is an IMU model without sensor fusion algorithms. It will therefore not produce estimates of orientation (quaternion/Euler/rotation matrix) or Free Acceleration. This also means that Rviz cannot show the real-time orientation of your MTi-100. Other outputs such as angular_velocity and acceleration are available.

Taeyoung96, 2022-09-30:
@StevenXsens   
Thank you for your help.  
However, other problems continue to arise. :cry:   

I fix my Device Settings and it could show data on MT manager like below.  

<p align="center">[image]</p>  

<p align="center">[image]</p>  

And I execute `roslaunch xsens_mti_driver xsens_mti_node.launch`, and the results are shown below.  
(Because I didn't have `orientation data`, the "display example" didn't work for me.  

```
roslaunch xsens_mti_driver xsens_mti_node.launch 
... logging to /home/taeyoung/.ros/log/d8eae8d2-3f34-11ed-877f-3497f6927eaa/roslaunch-taeyoung-cilab-23885.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://taeyoung-cilab:37337/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /xsens_mti_node/baudrate: 115200
 * /xsens_mti_node/device_id: 017828BD
 * /xsens_mti_node/frame_id: imu_link
 * /xsens_mti_node/port: /dev/ttyUSB0
 * /xsens_mti_node/pub_acceleration: True
 * /xsens_mti_node/pub_angular_velocity: True
 * /xsens_mti_node/pub_dq: True
 * /xsens_mti_node/pub_dv: True
 * /xsens_mti_node/pub_free_acceleration: False
 * /xsens_mti_node/pub_gnss: False
 * /xsens_mti_node/pub_imu: True
 * /xsens_mti_node/pub_mag: True
 * /xsens_mti_node/pub_positionLLA: False
 * /xsens_mti_node/pub_pressure: True
 * /xsens_mti_node/pub_quaternion: True
 * /xsens_mti_node/pub_sampletime: True
 * /xsens_mti_node/pub_temperature: True
 * /xsens_mti_node/pub_transform: True
 * /xsens_mti_node/pub_twist: False
 * /xsens_mti_node/pub_velocity: False
 * /xsens_mti_node/publisher_queue_size: 5

NODES
  /
    xsens_mti_node (xsens_mti_driver/xsens_mti_node)

ROS_MASTER_URI=http://localhost:11311

process[xsens_mti_node-1]: started with pid [23903]
[ INFO] [1664502776.199169032]: Creating XsControl object...
[ INFO] [1664502776.212026752]: Found baudrate parameter: 115200
[ INFO] [1664502776.212706261]: Found device ID parameter: 017828BD.
[ INFO] [1664502776.213331003]: Found port name parameter: /dev/ttyUSB0
[ INFO] [1664502776.213350274]: Scanning port /dev/ttyUSB0 ...
[ INFO] [1664502776.245586543]: Found a device with ID: 017828BD @ port: /dev/ttyUSB0, baudrate: 115200
[ INFO] [1664502776.245684711]: Opening port /dev/ttyUSB0 ...
[ INFO] [1664502776.301558916]: Device: MTi-100-2A8G4, with ID: 017828BD opened.
[ INFO] [1664502776.332867440]: Measuring ...
```

But I didn't get any data from this launch file.  
```
taeyoung@taeyoung-cilab:~/catkin_ws$ rostopic list
/filter/quaternion
/imu/acceleration
/imu/angular_velocity
/imu/data
/imu/dq
/imu/dv
/imu/mag
/imu/time_ref
/pressure
/rosout
/rosout_agg
/temperature
/tf
taeyoung@taeyoung-cilab:~/catkin_ws$ rostopic hz /imu/data 
subscribed to [/imu/data]
no new messages
no new messages
no new messages
no new messages
no new messages
^Cno new messages
taeyoung@taeyoung-cilab:~/catkin_ws$ rostopic hz /imu/acceleration 
subscribed to [/imu/acceleration]
no new messages
no new messages
no new messages
no new messages
no new messages
^Cno new messages
taeyoung@taeyoung-cilab:~/catkin_ws$ rostopic hz /imu/angular_velocity 
subscribed to [/imu/angular_velocity]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
^Cno new messages
```

According to Xsens Mti driver Troubleshooting, I also check `setserial` command.  
I got following error.  
```
taeyoung@taeyoung-cilab:~/catkin_ws$ setserial /dev/ttyUSB0 low_latency
Cannot get serial info: Inappropriate ioctl for device
```

I'd appreciate it if you could give me an answer.

Thanks,

StevenXsens, 2022-09-30:
Hi @Taeyoung96,

At this point your MTi is still not configured to output the right data:

- /imu/acceleration is only published if the MTi outputs the Acceleration message
- /imu/angular_velocity is only published if the MTi outputs the RateOfTurn message
- /imu/data is only published if the MTi outputs at least one of: Acceleration, RateOfTurn or Orientation

However with your current output configuration I think you should already be able to receive:
/imu/dq
/imu/dv
/imu/mag
/imu/time_ref
/pressure
/temperature

It might beconfusing that MT Manager is able to plot the Acceleration/RateOfTurn data. However, this is because MT Manager is able to convert the dv and dq messages into Acceleration/RateOfTurn data. The ROS driver does not support this.

Taeyoung96, 2022-09-30:
@StevenXsens   
Thanks for your help.  
Now I could get `/imu/data` normally! :+1: 
```
---
header: 
  seq: 3194
  stamp: 
    secs: 1664536000
    nsecs: 620643238
  frame_id: "imu_link"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 3.57851313311e-05
  y: 0.00446774158627
  z: -0.000764021475334
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration: 
  x: 0.0245991684496
  y: 0.0108192535117
  z: 9.82582092285
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

closed 2022-09-30 by Taeyoung96